eugene-tarassov/vivado-risc-v #99: USB port

jack4041313, 2022-08-15:
I want to add USB port on genesys2
Vivado axi usb 2.0 IP cannot be used because the pins do not match (input & output)
Do you have a suitable USB IP and driver for genesys2?

eugene-tarassov, 2022-08-15:
> Vivado axi usb 2.0 IP cannot be used because the pins do not match (input & output)

Both Vivado and Genesys2 use ULPI interface. Pins appear to match.
I tried to add USB 2.0 IP to the block design - it worked fine.
However, the IP requires purchasing a license.
I'm not aware of a free USB 2.0 core.

jack4041313, 2022-08-16:
Yes you are right
add USB 2.0 IP to the block design is worked fine
in vivado, USB 2.0 IP ULPI interface data section is separate
input data and output data is not the same pin
but on genesys2 ULPI interface,input data and output data is combined(inout)
this is my confusion point when writing constrain file

eugene-tarassov, 2022-08-16:
> input data and output data is not the same pin
> but on genesys2 ULPI interface,input data and output data is combined(inout)

Vivado automatically adds IOBUFs in the design wrapper to convert I, O and T pins to IO pins when UPLI is made external.

jack4041313, 2022-08-16:
Do you mean riscv_wrapper will combine these 3 pins automatically?
This is my block design, I have executed synthesis, but these three pins are not merged in my riscv_wrapper
[image]
[image]

jack4041313, 2022-08-17:
oh I should create a interface port 
Thank you for your response !

closed 2022-08-22 by jack4041313