eugene-tarassov/vivado-risc-v #107: no targets found with "name =~ "Hart #0*"".

yswntht, 2022-09-12:
Hi @eugene-tarassov , 

I used nexys-video as base template and ported to a new fpga "xc7k160t". I see timing is met with system clock at 50MHz and bit stream is successfully generated. my board uses PUSH-buttons for reset (button press connects to logic 1) and doenn't have an sdcard. To match the system reset i changed reset polarity to ACTIVE_HIGH in riscv.2020.1.tcl. To program the board I followed your suggestion from previous threads (using xsdb). I see following message:
```

****** Xilinx System Debugger (XSDB) v2020.1
  **** Build date : May 27 2020-20:33:44
    ** Copyright 1986-2020 Xilinx, Inc. All Rights Reserved.

xsdb% connect                                                                   
tcfchan#0                                                                       
xsdb% targets -set -filter {name =~ "Hart #0*"}                                 
no targets found with "name =~ "Hart #0*"". available targets:                  
  1  xc7k160t
     2  RISC-V JTAG DTM (Debug Transport Module is held in reset)

```
Any pointers what could be the issue?

Thanks.

eugene-tarassov, 2022-09-12:
> To match the system reset i changed reset polarity to ACTIVE_HIGH in riscv.2020.1.tcl

You also need to remove inverter (e.g. util_vector_logic_0) on the reset input.

yswntht, 2022-09-14:
Thank a lot. Your suggestion worked. I see the correct functionality now. 

```
****** Xilinx System Debugger (XSDB) v2020.1
  **** Build date : May 27 2020-20:33:44
    ** Copyright 1986-2020 Xilinx, Inc. All Rights Reserved.

xsdb% connect                                                                   
tcfchan#0                                                                       
xsdb% ta                                                                        
  1  xc7k160t
     2  RISC-V
        3  Hart #0 (Running)
xsdb% exit                                                                      
exit
```

closed 2024-02-13 by eugene-tarassov