eugene-tarassov/vivado-risc-v #113: build /u-boot/arch/riscv/cpu/start.o failed in ubuntu18.04

Yuxin-Yu, 2022-09-27:
Hello, I used the latest project, and the execution error was reported in Ubuntu18.04
[image]
Do you known how can I fix it ?

eugene-tarassov, 2022-09-27:
It works fine for me with Ubuntu 20.04.5 LTS.

Yuxin-Yu, 2022-09-29:
> This error is related to SSL library version. You, probably, just need to update your system with latest packages.
Thank you sir.But sometime when I execute "./mk-sd-image",there are  another error
[image]
How Can fix it,thanks

Yuxin-Yu, 2022-09-29:
I have solved this tag error.It is related to linux-stable version

closed 2022-09-29 by Yuxin-Yu