eugene-tarassov/vivado-risc-v #114: can not use g++

user10724, 2022-10-04:
Thank you so much for your open source project, it's very interesting. I am new to FPGA. I'm very sorry if the question is simple. But I have some problems compiling a C++ "hello world" program on a debian system. It reported the following error " -bash: g++: command not find" .It makes me confused，can you help me please！Thanks a lot.

```
debian@debian:~/cppfile$ g++ a.cpp -o a
-bash: g++: command not found
```
 I use g tab tab and really don't have g++. I use Genesys2, the parameter is rocket64b2gem. There may be some problems with my ethernet, not sure if this is the cause. Also, I replaced /etc/apt/sources.list.

```
debian@debian:~/cppfile$ ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN group default qlen 1000
    link/ether 00:0a:35:00:00:02 brd ff:ff:ff:ff:ff:ff
3: sit0@NONE: <NOARP> mtu 1480 qdisc noop state DOWN group default qlen 1000
    link/sit 0.0.0.0 brd 0.0.0.0
```
I tried updating g++ but it didn't work.
```
debian@debian:~/cppfile$ sudo apt-get install g++
sudo: unable to resolve host debian: Temporary failure in name resolution
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'libgpg-error-l10n' for regex 'g+'
Note, selecting 'dpkg' for regex 'g+'
Note, selecting 'default-logind' for regex 'g+'
Note, selecting 'libsasl2-modules-gssapi-mit' for regex 'g+'
Note, selecting 'rsyslog-pgsql' for regex 'g+'
Note, selecting 'base-config' for regex 'g+'
Note, selecting 'manpages-fr-extra' for regex 'g+'
Note, selecting 'moonshot-gss-eap' for regex 'g+'
Note, selecting 'libgssapi-krb5-2' for regex 'g+'
Note, selecting 'debian-ports-archive-keyring' for regex 'g+'
Note, selecting 'iputils-ping' for regex 'g+'
Note, selecting 'manpages-de' for regex 'g+'
Note, selecting 'manpages-fr' for regex 'g+'
Note, selecting 'manpages-pl' for regex 'g+'
Note, selecting 'libpam0g-dev' for regex 'g+'
Note, selecting 'rsyslog-openssl' for regex 'g+'
Note, selecting 'system-log-daemon' for regex 'g+'
Note, selecting 'dialog' for regex 'g+'
Note, selecting 'gpm' for regex 'g+'
Note, selecting 'libunistring2' for regex 'g+'
Note, selecting 'ping' for regex 'g+'
Note, selecting 'libmp3-tag-perl' for regex 'g+'
Note, selecting 'libgtk3-perl' for regex 'g+'
Note, selecting 'rsyslog-gssapi' for regex 'g+'
Note, selecting 'libpam0g-util' for regex 'g+'
Note, selecting 'e2fsprogs-l10n' for regex 'g+'
Note, selecting 'ctags' for regex 'g+'
Note, selecting 'libgcc1' for regex 'g+'
Note, selecting 'rsyslog-mysql' for regex 'g+'
Note, selecting 'libapt-pkg5.0' for regex 'g+'
Note, selecting 'libapt-pkg6.0' for regex 'g+'
Note, selecting 'lighttpd' for regex 'g+'
Note, selecting 'molly-guard' for regex 'g+'
Note, selecting 'vim-gtk3' for regex 'g+'
Note, selecting 'mailagent' for regex 'g+'
Note, selecting 'linux-image-5.16.0-4-riscv64' for regex 'g+'
Note, selecting 'libapt-pkg' for regex 'g+'
Note, selecting 'libgpm2' for regex 'g+'
Note, selecting 'libmath-prime-util-gmp-perl' for regex 'g+'
Note, selecting 'logcheck-database' for regex 'g+'
Note, selecting 'gnupg1' for regex 'g+'
Note, selecting 'gnupg2' for regex 'g+'
[... 159 lines cut ...]
SwapTotal:        262140 kB
SwapFree:         262140 kB
Dirty:               176 kB
Writeback:             0 kB
AnonPages:         11024 kB
Mapped:            18576 kB
Shmem:              1240 kB
KReclaimable:       7036 kB
Slab:              15172 kB
SReclaimable:       7036 kB
SUnreclaim:         8136 kB
KernelStack:        1056 kB
PageTables:          620 kB
NFS_Unstable:          0 kB
Bounce:                0 kB
WritebackTmp:          0 kB
CommitLimit:      761016 kB
Committed_AS:     124176 kB
VmallocTotal:   67108863 kB
VmallocUsed:         216 kB
VmallocChunk:          0 kB
Percpu:              152 kB
debian@debian:/usr/share/gcc$ cat /proc/version
Linux version 5.16.14-dirty (eugene@gate) (riscv64-linux-gnu-gcc (Ubuntu 9.4.0-1ubuntu1~20.04) 9.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1 SMP Sat Mar 19 10:50:59 PDT 2022
```

eugene-tarassov, 2022-10-04:
> I use Genesys2

Good choice.

> the parameter is rocket64b2gem

I don't recommend  'gem' variants - unless you are a Gemmini developer.
Gemmini hardware is slow and not 100% stable yet, rocket64b2 works much better.

> There may be some problems with my ethernet

Yes, there is a problem. Your Ethernet is not working - according to 'ip a' output.

>  not sure if this is the cause

Yes, it is the cause - you need working Ethernet and Internet access to install g++.

> Also, I replaced /etc/apt/sources.list.

Why? sources.list contains address of Debian repo, you need it to install g++.

With working Ethernet, after `apt update && apt upgrade`, `apt install g++` will work as expected:
```
root@debian:~# apt install g++
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following additional packages will be installed:
  g++-12 libstdc++-12-dev
Suggested packages:
  gcc-12-doc libstdc++-12-doc
The following NEW packages will be installed:
  g++ g++-12 libstdc++-12-dev
0 upgraded, 3 newly installed, 0 to remove and 0 not upgraded.
Need to get 12.5 MB of archives.
After this operation, 65.2 MB of additional disk space will be used.
Do you want to continue? [Y/n]
```

user10724, 2022-10-05:
Thank you so much for your help! I have rebuilt the project, using rocket64b2 and not changing the source. It was indeed the ethernet issue. I didn't plug in the network cable at first. Now I plugged in the network cable and connected it to the computer, but I found that the genesys2 could not ping the external network . My computer can ping genesys2 and log into debian using ssh.  My computer can ping the external network. I also know very little about networking.I need to learn some networking knowledge. My eth0 only shows ipv6, I manually configured ipv4.
```
debian@debian:~$ ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether 00:0a:35:00:00:02 brd ff:ff:ff:ff:ff:ff
    inet 172.24.126.60/17 brd 172.24.127.255 scope global eth0
       valid_lft forever preferred_lft forever
    inet6 fe80::20a:35ff:fe00:2/64 scope link
       valid_lft forever preferred_lft forever
3: sit0@NONE: <NOARP> mtu 1480 qdisc noop state DOWN group default qlen 1000
    link/sit 0.0.0.0 brd 0.0.0.0
debian@debian:~$ sudo ping 172.24.126.61
PING 172.24.126.61 (172.24.126.61) 56(84) bytes of data.
^C
--- 172.24.126.61 ping statistics ---
27 packets transmitted, 0 received, 100% packet loss, time 26607ms

debian@debian:~$ ping 8.8.8.8
debian@debian:~$ sudo ping 8.8.8.8
PING 8.8.8.8 (8.8.8.8) 56(84) bytes of data.
From 172.24.126.60 icmp_seq=1 Destination Host Unreachable
From 172.24.126.60 icmp_seq=2 Destination Host Unreachable
From 172.24.126.60 icmp_seq=3 Destination Host Unreachable
^C
--- 8.8.8.8 ping statistics ---
5 packets transmitted, 0 received, +3 errors, 100% packet loss, time 4070ms
pipe 4
debian@debian:~$ sudo ping locolhost
^C
debian@debian:~$ sudo ping localhost
PING localhost(localhost (::1)) 56 data bytes
64 bytes from localhost (::1): icmp_seq=1 ttl=64 time=0.981 ms
64 bytes from localhost (::1): icmp_seq=2 ttl=64 time=0.880 ms
64 bytes from localhost (::1): icmp_seq=3 ttl=64 time=1.30 ms
^C
--- localhost ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 2005ms
rtt min/avg/max/mdev = 0.880/1.055/1.304/0.180 ms
debian@debian:~$ sudo ping 127.24.126.59
PING 127.24.126.59 (127.24.126.59) 56(84) bytes of data.
64 bytes from 127.24.126.59: icmp_seq=1 ttl=64 time=0.821 ms
64 bytes from 127.24.126.59: icmp_seq=2 ttl=64 time=1.17 ms
64 bytes from 127.24.126.59: icmp_seq=3 ttl=64 time=0.728 ms
^C
--- 127.24.126.59 ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 2004ms
rtt min/avg/max/mdev = 0.728/0.905/1.168/0.189 ms
debian@debian:~$ sudo ping 127.24.0.1
PING 127.24.0.1 (127.24.0.1) 56(84) bytes of data.
64 bytes from 127.24.0.1: icmp_seq=1 ttl=64 time=0.821 ms
64 bytes from 127.24.0.1: icmp_seq=2 ttl=64 time=0.726 ms
64 bytes from 127.24.0.1: icmp_seq=3 ttl=64 time=1.22 ms
^C
--- 127.24.0.1 ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 2004ms
rtt min/avg/max/mdev = 0.726/0.923/1.224/0.215 ms
debian@debian:~$ sudo ping 127.24.126.61
PING 127.24.126.61 (127.24.126.61) 56(84) bytes of data.
64 bytes from 127.24.126.61: icmp_seq=1 ttl=64 time=0.835 ms
64 bytes from 127.24.126.61: icmp_seq=2 ttl=64 time=0.747 ms
64 bytes from 127.24.126.61: icmp_seq=3 ttl=64 time=0.739 ms
^C
--- 127.24.126.61 ping statistics ---
```

eugene-tarassov, 2022-10-05:
You need to connect the board to a router. The router will assign IPv4 address, do packet forwarding, address translation etc. You could setup your computer to act like a router, but it is complicated. Just connect the board directly to your router.

closed 2022-10-06 by user10724

Yuxin-Yu, 2023-03-20:
> Thank you very much! I have connected to the Internet now, and set the computer to router mode through what you said. I also successfully updated g++ and now it can run cpp programs.
> 
> ```
> debian@debian:~/cppfiles$ ls
> a.cpp
> debian@debian:~/cppfiles$ g++ a.cpp -o a
> debian@debian:~/cppfiles$ ls
> a  a.cpp
> debian@debian:~/cppfiles$ ./a
> i=0hello world!
> i=1hello world!
> i=2hello world!
> i=3hello world!
> i=4hello world!
> ```
> 
> I did it through windows network sharing. I also tried the bridge but it didn't work. I still replaced the sources.list when updating g++ because I'm in China.
> 
> ```
> deb [arch=riscv64] http://mirrors.aliyun.com/debian-ports unstable main
> 
> deb [arch=riscv64] http://mirrors.aliyun.com/debian-ports unreleased main
> ```
> 
> I also modified the file /etc/network/interfaces to make genesys2 and the computer on the same network segment.
> 
> ```
> auto eth0
> #iface eth0 inet dhcp
> iface eth0 inet static 
> address 192.168.137.11
> netmask 255.255.255.0
> gateway 192.168.137.1
> 
> dns-nameservers 8.8.8.8
> dns-nameservers 114.114.114.114
> ```
> 
> Thanks for your response again!

Hi,bro.I changed /etc/apt/source.list and /etc/network/interfaces file as you said because I in China too.Bus when I execute command apt update ,I got some error, like:

> root@debian:/etc/apt# apt update
> Ign:1 http://mirrors.aliyun.com/debian-ports unstable InRelease
> Ign:2 http://mirrors.aliyun.com/debian-ports unreleased InRelease
> Ign:1 http://mirrors.aliyun.com/debian-ports unstable InRelease
> Ign:2 http://mirrors.aliyun.com/debian-ports unreleased InRelease
> Ign:1 http://mirrors.aliyun.com/debian-ports unstable InRelease
> Ign:2 http://mirrors.aliyun.com/debian-ports unreleased InRelease
> Err:1 http://mirrors.aliyun.com/debian-ports unstable InRelease
>   Cannot initiate the connection to mirrors.aliyun.com:80 (2409:8c00:1:2:3::3fe). - connect (101: Network is unreachable) Cannot initiate the connection to mirrors.aliyun.com:80 (2409:8c00:1:2:3::3fd). - connect (101: Network is unreachable) Could not connect to mirrors.aliyun.com:80 (124.193.244.224), connection timed out Could not connect to mirrors.aliyun.com:80 (124.193.244.225), connection timed out Could not connect to mirrors.aliyun.com:80 (124.193.244.229), connection timed out Could not connect to mirrors.aliyun.com:80 (124.193.244.228), connection timed out Could not connect to mirrors.aliyun.com:80 (124.193.244.227), connection timed out Could not connect to mirrors.aliyun.com:80 (124.193.244.230), connection timed out Could not connect to mirrors.aliyun.com:80 (124.193.244.226), connection timed out Could not connect to mirrors.aliyun.com:80 (124.193.244.231), connection timed out
> Err:2 http://mirrors.aliyun.com/debian-ports unreleased InRelease
>   Cannot initiate the connection to mirrors.aliyun.com:80 (2409:8c00:1:2:3::3fe). - connect (101: Network is unreachable) Cannot initiate the connection to mirrors.aliyun.com:80 (2409:8c00:1:2:3::3fd). - connect (101: Network is unreachable)
> Reading package lists... Done
> Building dependency tree... Done
> Reading state information... Done
> All packages are up to date.
> W: Failed to fetch http://mirrors.aliyun.com/debian-ports/dists/unstable/InRelease  Cannot initiate the connection to mirrors.aliyun.com:80 (2409:8c00:1:2:3::3fe). - connect (101: Network is unreachable) Cannot initiate the connection to mirrors.aliyun.com:80 (2409:8c00:1:2:3::3fd). - connect (101: Network is unreachable) Could not connect to mirrors.aliyun.com:80 (124.193.244.224), connection timed out Could not connect to mirrors.aliyun.com:80 (124.193.244.225), connection timed out Could not connect to mirrors.aliyun.com:80 (124.193.244.229), connection timed out Could not connect to mirrors.aliyun.com:80 (124.193.244.228), connection timed out Could not connect to mirrors.aliyun.com:80 (124.193.244.227), connection timed out Could not connect to mirrors.aliyun.com:80 (124.193.244.230), connection timed out Could not connect to mirrors.aliyun.com:80 (124.193.244.226), connection timed out Could not connect to mirrors.aliyun.com:80 (124.193.244.231), connection timed out
> W: Failed to fetch http://mirrors.aliyun.com/debian-ports/dists/unreleased/InRelease  Cannot initiate the connection to mirrors.aliyun.com:80 (2409:8c00:1:2:3::3fe). - connect (101: Network is unreachable) Cannot initiate the connection to mirrors.aliyun.com:80 (2409:8c00:1:2:3::3fd). - connect (101: Network is unreachable)
> W: Some index files failed to download. They have been ignored, or old ones used instead.

Do you know how can I fix it,Thank you @user10724