eugene-tarassov/vivado-risc-v #115: “Prepare the SD card ” failed

StilluSmile, 2022-10-20:
The "prepare sd card" link fails when I connect my sd card and execute:

./mk-sd-card

Hint: Checking that no-one is using this disk right now ... FAILED

This disk is currently in use - repartitioning is probably a bad idea.
Umount all file systems, and swapoff all swap partitions on this disk.
Use the --no-reread flag to suppress this check.

How to fix this issue?  thanks for any suggestion and help.

StilluSmile, 2022-10-20:
> The "prepare sd card" link fails when I connect my sd card and execute:
>
> ./mk-sd-card
>
> Hint: Checking that no-one is using this disk right now ... FAILED
>
> This disk is currently in use - [...]

According to this prompt, I modified mk-sd-card and added --no-reread to sfdisk to suppress this check. However, it also prompts: Re-reading the partition table failed.: Device or resource busy. I wonder if It is executed correctly. The full output is logged as follows:

make: 'debian-riscv64/initrd' is up to date.
make: 'debian-riscv64/rootfs.tar.gz' is up to date.
1500+0 records in
1500+0 records out
1572864000 bytes (1.6 GB, 1.5 GiB) copied, 6.20882 s, 253 MB/s
SD image device: /dev/loop13
Checking that no-one is using this disk right now ... OK

Disk /dev/loop13: 1.48 GiB, 1572864000 bytes, 3072000 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes

>>> Created a new DOS disklabel with disk identifier 0x61981640.
/dev/loop13p1: Created a new partition 1 of type 'W95 FAT16 (LBA)' and of size 64 MiB.
/dev/loop13p2: Created a new partition 2 of type 'Linux' and of size 1.4 GiB.
/dev/loop13p3: Done.

New situation:
Disklabel type: dos
Disk identifier: 0x61981640

Device        Boot  Start     End Sectors  Size Id Type
/dev/loop13p1 *      2048  133119  131072   64M  e W95 FAT16 (LBA)
/dev/loop13p2      133120 3071999 2938880  1.4G 83 Linux

The partition table has been altered.
mkfs.fat 4.1 (2017-01-24)
mke2fs 1.45.5 (07-Jan-2020)
Creating filesystem with 367360 4k blocks and 91968 inodes
Filesystem UUID: 68d82fa1-1bb5-435f-a5e3-862176586eec
Superblock backups stored on blocks: 
	32768, 98304, 163840, 229376, 294912

Allocating group tables: done                            
Writing inode tables: done                            
Creating journal (8192 blocks): done
Writing superblocks and filesystem accounting information: done 

~/vivado-risc-v/debian-riscv64/rootfs ~/vivado-risc-v
~/vivado-risc-v
~/vivado-risc-v/debian-riscv64/boot ~/vivado-risc-v
~/vivado-risc-v

Boot partition:
Filesystem     1K-blocks  Used Available Use% Mounted on
/dev/loop13p1      65390 24260     41130  38% /home/buaaxdd/vivado-risc-v/debian-riscv64/boot
total 1290
-rwxr-xr-x 1 root root 1317664 10月 20 10:53 boot.elf
drwxr-xr-x 2 root root    2048 10月 20 10:53 extlinux

Root partition:
Filesystem     1K-blocks   Used Available Use% Mounted on
/dev/loop13p2    1409240 690504    628880  53% /home/buaaxdd/vivado-risc-v/debian-riscv64/rootfs
total 262232
drwxr-xr-x  2 root root      4096 3月  19  2022 bin
drwxr-xr-x  2 root root      4096 3月  19  2022 boot
drwxr-xr-x  4 root root      4096 1月  20  2022 dev
drwxr-xr-x 62 root root      4096 3月  19  2022 etc
drwxr-xr-x  3 root root      4096 3月  13  2022 home
drwxr-xr-x 11 root root      4096 3月  19  2022 lib
drwx------  2 root root     16384 3月  13  2022 lost+found
drwxr-xr-x  2 root root      4096 3月  13  2022 media
drwxr-xr-x  2 root root      4096 3月  13  2022 mnt
drwxr-xr-x  2 root root      4096 3月  13  2022 opt
drwxr-xr-x  2 root root      4096 1月  20  2022 proc
drwx------  3 root root      4096 3月  13  2022 root
drwxr-xr-x  4 root root      4096 3月  13  2022 run
drwxr-xr-x  2 root root      4096 3月  19  2022 sbin
drwxr-xr-x  2 root root      4096 3月  13  2022 srv
-rw-------  1 root root 268435456 3月  19  2022 swapfile
drwxr-xr-x  2 root root      4096 1月  20  2022 sys
drwxrwxrwt  2 root root      4096 3月  19  2022 tmp
drwxr-xr-x 11 root root      4096 3月  13  2022 usr
drwxr-xr-x 11 root root      4096 3月  13  2022 var

fsck.fat 4.1 (2017-01-24)
/dev/loop13p1: 6 files, 12130/32695 clusters
rootfs: 15972/91968 files (0.0% non-contiguous), 187677/367360 blocks

Copy disk image to:
Disk /dev/sdb: 29.12 GiB, 31267487744 bytes, 61069312 sectors
Disk model: SD/MMC          
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x7166187a

Device     Boot  Start     End Sectors  Size Id Type
/dev/sdb1  *      2048  133119  131072   64M  e W95 FAT16 (LBA)
/dev/sdb2       133120 3071999 2938880  1.4G 83 Linux
Are you sure? [y/N] y
1563492352 bytes (1.6 GB, 1.5 GiB) copied, 28 s, 55.8 MB/s
24000+0 records in
24000+0 records out
1572864000 bytes (1.6 GB, 1.5 GiB) copied, 28.522 s, 55.1 MB/s
Disk /dev/sdb: 29.12 GiB, 31267487744 bytes, 61069312 sectors
Disk model: SD/MMC          
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x61981640

Old situation:

Device     Boot  Start     End Sectors  Size Id Type
/dev/sdb1  *      2048  133119  131072   64M  e W95 FAT16 (LBA)
/dev/sdb2       133120 3071999 2938880  1.4G 83 Linux

/dev/sdb2: 
New situation:
Disklabel type: dos
Disk identifier: 0x61981640

Device     Boot  Start      End  Sectors  Size Id Type
/dev/sdb1  *      2048   133119   131072   64M  e W95 FAT16 (LBA)
/dev/sdb2       133120 61069311 60936192 29.1G 83 Linux

The partition table has been altered.
Calling ioctl() to re-read partition table.
**Re-reading the partition table failed.: Device or resource busy
The kernel still uses the old table. The new table will be used at the next reboot or after you run partprobe(8) or kpartx(8).**
Syncing disks.
fsck from util-linux 2.34
e2fsck 1.45.5 (07-Jan-2020)
Pass 1: Checking inodes, blocks, and sizes
Pass 2: Checking directory structure
Pass 3: Checking directory connectivity
Pass 4: Checking reference counts
Pass 5: Checking group summary information
rootfs: 15972/91968 files (0.0% non-contiguous), 187677/367360 blocks
resize2fs 1.45.5 (07-Jan-2020)
Resizing the filesystem on /dev/sdb2 to 7617024 (4k) blocks.
The filesystem on /dev/sdb2 is now 7617024 (4k) blocks long.

eugene-tarassov, 2022-10-20:
I have seen this on some versions of Linux, and it looks like a Linux bug.
Anyway, the script does run partprobe, and it looks like it worked OK.
I think you can ignore the message.

closed 2023-05-30 by eugene-tarassov