eugene-tarassov/vivado-risc-v #116: PLL clock setting

JackChenggg, 2022-10-29:
If I change the PLL settings, such as adding a clock, do I need to change the device tree settings?

eugene-tarassov, 2022-10-29:
No, you don't need to change the device tree after adding a clock.
Unless you plan to add a device driver that needs info about that clock in the device tree.

JackChenggg, 2022-10-29:
If "video test pattern generator" is used
Device tree needs to give "clocks" information
I added clk_out5 to the PLL,
What information do I need to add to the device tree?
[image]
[image]

eugene-tarassov, 2022-10-29:
Yes, for TPG driver you need to define a clock in the board bootrom.dts.
Refer to the clock provider schema: [clock.yaml](https://github.com/devicetree-org/dt-schema/blob/main/dtschema/schemas/clock/clock.yaml)

Something like:
```
	clocks {
		#address-cells = <1>;
		#size-cells = <1>;
		ranges;

		clk_out5: clk5 {
			compatible = "fixed-clock";
			#clock-cells = <0>;
			clock-frequency = <250000000>;
		};
	}
```
Then use it like:
```
    clocks = <&clk_out5 0>;
```

closed 2023-05-30 by eugene-tarassov