eugene-tarassov/vivado-risc-v #118: Can you point me how to download the relate resources?

e665107, 2022-11-15:
hi
When i execute the next cmd
make update-submodules
it stop at the followling log

Cloning into '/home/e665106/Downloads/vivado-risc-v/linux-stable'...

e665107, 2022-11-16:
Understand, I use ubuntu20.04, which distruibution you use?  And do you have a fast source link?

eugene-tarassov, 2022-11-16:
> I use ubuntu20.04

Yes, this is the recommended version to run this repo

> And do you have a fast source link?

The link in .gitmodules points to the Linux source origin at kernel.org
It works fast for me - most of the time. You can try to change it to a mirror that works better for your location.

e665107, 2022-11-16:
hi 
 If i use the git clone with the link of .gitmodule, it can be downloaded! I don't know why.
git clone https://github.com/ucb-bar/gemmini
Cloning into 'gemmini'...
remote: Enumerating objects: 11327, done.        
remote: Counting objects: 100% (2886/2886), done.        
remote: Compressing objects: 100% (521/521), done.        
remote: Total 11327 (delta 2301), reused 2713 (delta 2172), pack-reused 8441        
Receiving objects: 100% (11327/11327), 3.76 MiB | 4.40 MiB/s, done.
Resolving deltas: 100% (5879/5879), done.

e665107, 2022-11-17:
Hi 
 I find a question, when i execute the followling cmd.
 I clone the newest verilog-ethernet repository. So i comment this patch, and others have passed
e665106@ubuntu:~/Downloads/vivado-risc-v$ make CONFIG=rocket32s1 BOARD=kc705 bitstream
if [ -s patches/ethernet.patch ] ; then cd ethernet/verilog-ethernet && ( git apply -R --check ../../patches/ethernet.patch 2>/dev/null || git apply ../../patches/ethernet.patch ) ; fi
error: patch failed: rtl/eth_phy_10g_rx_ber_mon.v:59
error: rtl/eth_phy_10g_rx_ber_mon.v: patch does not apply
error: patch failed: rtl/eth_phy_10g_rx_watchdog.v:68
error: rtl/eth_phy_10g_rx_watchdog.v: patch does not apply
make: *** [Makefile:186: hdl-patch] Error 1

e665107, 2022-11-17:
Hi
I find a new problem! 
I also clone the newest rocket-chip repository.
[error] /home/e665106/Downloads/vivado-risc-v/rocket-chip/src/main/scala/package.scala:6:16: not found: value chipsalliance

eugene-tarassov, 2022-11-17:
Yue cannot use arbitrary versions of submodules. The code will either not build or not work. By default, the repo checkouts versions that are tested and known to work. Don't change submodule versions unless you have a very good reason for that.

e665107, 2022-11-17:
Understand,  I use the link in .gitmodule and clone directly!   If it works well with the readme operation, i would not do that!  Franky speaking,  i am unfamiar with the scala. So I have get encountered many questions. And problems above have been overcomed.
And now the new question is following:

[error] /home/e665106/Downloads/vivado-risc-v/generators/gemmini/src/main/scala/gemmini/TransposePreloadUnroller.scala:8:8: not found: object midas
I think  the reason is that the relate lib named midas is loss. Can you give me a hand?  Thank you very much!

eugene-tarassov, 2022-11-19:
You have to use exact same version of submodules as this repo is pointing to. This is cutting edge of RISC-V development, changes in submodules are coming in daily, many of them break the build. The easiest way to get it working is to delete whatever you have got after unsuccessful download and start from scratch according to the instructions. Network problems are temporary, it should work eventually.

closed 2023-05-30 by eugene-tarassov