eugene-tarassov/vivado-risc-v #119: VC 707 Flash Module is Broken

gnodipac886, 2022-11-16:
Hi all,

My flash memory chip on my vc 707 is fault, and does not even pass the BIST test. Is there a way of bypassing the flash stage and just program the bit stream via the jtag?

Thanks :)

eugene-tarassov, 2022-11-17:
Yes.
Vivado, Vitis and XSDB all have commands to program bitstream via JTAG.
For example, in XSDB:
```
connect
fpga workspace/rocket64b2/vivado-vc707-riscv/vc707-riscv.runs/impl_1/riscv_wrapper.bit
```

gnodipac886, 2022-11-17:
@eugene-tarassov  I would also like to ask whether ethernet support for VC707 already? I see the ethernet files under boards/vc707 but when I boot it up and do ifconfig, I only get localhost 127.0.0.1

gnodipac886, 2022-11-17:
currently it shows:
```root@debian:~# ip addr show
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 00:0a:35:00:00:00 brd ff:ff:ff:ff:ff:ff
3: sit0@NONE: <NOARP> mtu 1480 qdisc noop state DOWN group default qlen 1000
    link/sit 0.0.0.0 brd 0.0.0.0```

eugene-tarassov, 2022-11-17:
Yes, Ethernet works on VC707.

gnodipac886, 2022-11-24:
Happy Thanksgivng!

> Right, they are not synthesized optimally, use a lot of LUTs. Bigger mesh would not fit into FPGA. At least, it used to be the case. I have not tried bigger mesh with the latest Gemmini source.

I wrote the processing element (PE) in verilog and now I can generate a 4x4 one :)
however, I still can't generate a 8x8 one due to lack of resources. In the image below, you can see that its actually the scratch pad that is consuming the most LUTs. VC707 has a total of around 300K LUTs

Would it be possible if we tried to infer block ram? I'm not sure how that would work. It seems like Chisel really generates inefficient verilog code -.-

[image]

eugene-tarassov, 2022-11-26:
> scratch pad ... is consuming the most LUTs

This is not normal, and I cannot reproduce.
The scratch pad uses block RAM and looks fine when I try to build Rocket64b2gem:

[image]

gnodipac886, 2022-11-27:
> This is not normal, and I cannot reproduce.
> The scratch pad uses block RAM and looks fine when I try to build Rocket64b2gem:

I think I found the issue. I was wondering if you can help me find a fix for this repo?
The bug resides in how sram.v is generated. The following code is for a 16x16 gemmini config
In the repo, the mem is generated by declaring a single register cluster "ram" as such 

```
module mem_0_ext_16(
  input W0_clk,
  input [8:0] W0_addr,
  input W0_en,
  input [511:0] W0_data,
  input [63:0] W0_mask,
  input R0_clk,
  input [8:0] R0_addr,
  input R0_en,
  output [511:0] R0_data
);

  reg reg_R0_ren;
  reg [8:0] reg_R0_addr;
  reg [511:0] ram [511:0];
  ...
```

while the mem that gemmini would generate using their build_vcs.sh script under gemmini/scripts generates several chunk of split_mem as follows:

```
module mem_0_ext(
  input  [8:0]   W0_addr,
  input          W0_clk,
  input  [511:0] W0_data,
  input          W0_en,
  input  [63:0]  W0_mask,
  input  [8:0]   R0_addr,
  input          R0_clk,
  output [511:0] R0_data,
  input          R0_en
);
...
split_mem_0_ext mem_0_0 (
    .W0_addr(mem_0_0_W0_addr),
    .W0_clk(mem_0_0_W0_clk),
    .W0_data(mem_0_0_W0_data),
    .W0_en(mem_0_0_W0_en),
    .W0_mask(mem_0_0_W0_mask),
    .R0_addr(mem_0_0_R0_addr),
    .R0_clk(mem_0_0_R0_clk),
    .R0_data(mem_0_0_R0_data),
    .R0_en(mem_0_0_R0_en)
  );
...
  split_mem_0_ext mem_0_63 (
    .W0_addr(mem_0_63_W0_addr),
    .W0_clk(mem_0_63_W0_clk),
    .W0_data(mem_0_63_W0_data),
    .W0_en(mem_0_63_W0_en),
    .W0_mask(mem_0_63_W0_mask),
    .R0_addr(mem_0_63_R0_addr),
    .R0_clk(mem_0_63_R0_clk),
    .R0_data(mem_0_63_R0_data),
    .R0_en(mem_0_63_R0_en)
  );
...
```

Using this split mem design, BRAM was inferred successfully and now the 16x16 gemmini deisgn looks like this
[image]

I have also attached the files to the different sram files as well (had to do .txt since github doesn't do .v files...)
[sram_16_from_repo.txt](https://github.com/eugene-tarassov/vivado-risc-v/files/10097634/sram_16_from_repo.txt)
[sram_gemmini_generated.txt](https://github.com/eugene-tarassov/vivado-risc-v/files/10097635/sram_gemmini_generated.txt)

eugene-tarassov, 2022-11-28:
> Using this split mem design, BRAM was inferred successfully...

Interesting. I will investigate.

> Do you happen to know if it's possible to swap the dram on these fpga boards?

I believe it is possible. To make it work, you need changes in the FPGA design:
1. Use Rocket Chip configuration with at least 34-bit memory address bus. By default, it is 32-bit, not enough to address 8GB. You need to add `new WithExtMemSize(0x200000000L) ++`, see Rocket64b2m config as example.
2. Rebuild the project with MEMORY_SIZE argument `make MEMORY_SIZE=0x200000000 BOARD=vc707 ...`.
3. In Vivado block design, re-configure DDR3 controller for the new memory module.

gnodipac886, 2022-11-28:
> Interesting. I will investigate.
> 
> >

After some initial investigation, there are some timing issues when I build with split_mem... Will report back though

I also built Gemmini on VCS with 2x2 configuration, same as this project, and running basic examples doesn't seem to work even in VCS simulation. Most likely due to the small sys array size. I think it first starts working at 8x8 or 16x16. If we can get the split_mem to work, we don't need to worry about this :)

gnodipac886, 2022-11-29:
> 3\. In Vivado block design, re-configure DDR3 controller for the new memory module.

A bit confused as to how this step works, could you point me to the right direction? Thank you!

eugene-tarassov, 2022-11-29:
In Vivado, click Open Block Design, double click DDR block in the diagram, double click mig_7series_0 - it will open re-configuration dialog box. Look through pages of the dialog box, there is Memory Part selection.

gnodipac886, 2022-11-29:
> In Vivado, click Open Block Design, double click DDR block in the diagram, double click mig_7series_0 - it will open re-configuration dialog box. Look through pages of the dialog box, there is Memory Part selection.

That almost worked, it seems like it is write locked somehow? It can only be generated by the parent project? See error message below.

How's the BRAM going? I'd be happy to help you out!

[image]

eugene-tarassov, 2022-11-29:
> That almost worked, it seems like it is write locked somehow?

It looks like a glitch in Vivado.

> have directly edited riscv-2021.1.tcl ...

Yes, eventually you have to edit the Tcl file.
But I would use Vivado GUI to validate and test the changes before editing the file.

> How's the BRAM going?

I tried all kind of tests, and Vivado appears to fail to infer block RAM when a write mask is used. It looks like a bug in Vivado. I modified the script to split RAM to slices along write mask. It works much better. I will commit after some more testing.

gnodipac886, 2022-11-29:
> I tried all kind of tests, and Vivado appears to fail to infer block RAM when a write mask is used. It looks like a bug in Vivado. I modified the script to split RAM to slices along write mask. It works much better. I will commit after some more testing.

Would it be possible to have your current updates on a separate branch? I have a project report tomorrow and would love to have some results here :)

Also [Gemmini](https://github.com/ucb-bar/gemmini) has had a version bump to 0.7.0, could be worth while investigating. In their commit message, they claim to have power and area reductions.

Lastly, I have added a blackbox PE design using systemverilog in [my fork of gemmini](https://github.com/gnodipac886/gemmini/tree/PE_BLACKBOX). The SV design is [here](https://github.com/gnodipac886/gemmini/blob/PE_BLACKBOX/src/main/resources/vsrc/full_mul.sv). It should have significant area savings. Chisel synthesized 700+ LUTs while my design only takes around 325 LUTs. In order to include the design in the build you need to add the file in vivado.tcl which I have [here](https://github.com/gnodipac886/vivado-risc-v/blob/master/vivado.tcl) in my fork of your repo.

Gemmini's original PE unit **might** have synthesized 2 MAC units (didn't check, just guessing) in one PE unit while mine uses just 1. Using my MAC unit, I was able to synthesize a 16x16 design (though with timing issues caused by spad which you will likely fix in your update). I cannot synthesize(vivado implement) a 16x16 design with their original design (309K+ LUTs). 

Could you test with these? Or if you add the updates I can try testing them with my design :)

Cheers mate

eugene-tarassov, 2022-11-30:
> Would it be possible to have your current updates on a separate branch?

The tests are running OK so far, so I have committed the changes to the master branch.
I test RocketChip itself, I don't run Gemmini functional tests, only check that it builds.
8x8 looks OK, 16x16 is about 15K LUTs bigger than VC707 FPGA size.
I'll check the latest Gemmini version and your fork.

gnodipac886, 2022-11-30:
> I'll check the latest Gemmini version and your fork.

I just fixed my fork of Gemmini, originally the pushed version was using the original PE design.

gnodipac886, 2022-11-30:
> 8x8 looks OK, 16x16 is about 15K LUTs bigger than VC707 FPGA size.

I just synthesized a 16x16 one with your newest update with my MAC unit and still getting timinig issues.
[image]
Were you able to compile without timing issues?
I will try to make a 8x8 one and see if there are timing issues.

eugene-tarassov, 2022-11-30:
Max clock frequency of Gemmini is rather low, about 30MHz. The Makefile selects clock frequency from this table [board/rocket-freq](https://github.com/eugene-tarassov/vivado-risc-v/blob/master/board/rocket-freq), but your configuration name Rocket64b2gem_custom does not have an entry in the table, so it uses default 100MHz - too high for Gemmini.

You can change the frequency without re-creating the project - double-click Clocking Wizard in the Block Design diagram.

gnodipac886, 2022-11-30:
> You can change the frequency without re-creating the project - double-click Clocking Wizard in the Block Design diagram.

Thank you for this info, will be reporting back :)

One more thing, I tried to add 8gb ram in the design by changing the tcl file, but I'm getting a complaint that says with current design I can only achieve 400Mhz clock which is not enough for the 8gb stick which needs a min of 800Mhz. It tells me to change the parameters etc. Do you know where I should look into?

Thanks for all this help by the way, really appreciate it! :)

eugene-tarassov, 2022-11-30:
> I tried to add 8gb ram ... It tells me to change the parameters etc. Do you know where I should look into?

No, I don't know. A common way to change the parameters is to use Memory Interface Generator dialog box. If it does not works, it becomes complicated.

gnodipac886, 2022-12-04:
> No, I don't know. A common way to change the parameters is to use Memory Interface Generator dialog box. If it does not works, it becomes complicated.

Currently, I have 8gb working on a standalone MIG build and I can see via ILA that it is reading correctly. This is based off the vc707 mig example on Xilinx website. However, I can't reproduce this result on this project, so I was wondering where I should add an ILA to monitor traffic in this project? (the AXI bus or the DDR output pins?)

> Yes, Ethernet works on VC707.

I realized that my debian images has been booting in emergency mode which could be the cause of ethernet being down on boot (ssh also disabled). Have you ever ran into an issue like this?

The debian image I've been using was originally built from mk-sd-image, then I booted this image using qemu to install the dependencies faster. I think this method is much faster than installing everything on the board. Could we add such a flow to this project? It would be very nice to have!

eugene-tarassov, 2022-12-04:
> I have 8gb working on a standalone MIG build

8GB DDR3 SODIMM is dual rank module. I found that support of dual rank memory was disabled in the Memory Interface Generator for VC707, because the board design does not meet timing requirements of dual rank SODIMM. I'm testing changes to support 4GB single rank modules. It looks OK so far.

> I realized that my debian images has been booting in emergency mode

I reproduced it using slow RISC-V config, 1 core at 50MHz. The problem is caused by udev daemon taking very long time to start, about 50min. systemd hits multiple timeouts, and OS falls into emergency mode. This used to work, and it works OK on faster configs. It looks like a bug in the latest Linux code, in either udev or kernel. As a workaround, I changed systemd timeouts to 2 hours, it works OK, but takes over 1 hour to boot. This is not normal. I'm not sure how to fix it yet.

> qemu to install the dependencies faster

If you post details on how you boot qemu, I will add this flow to the project.
I mostly use 64b2 100MHz config, it is pretty fast, so installing packages is not a problem.

gnodipac886, 2022-12-05:
> 8GB DDR3 SODIMM is dual rank module. I found that support of dual rank memory was disabled in the Memory Interface Generator for VC707, because the board design does not meet timing requirements of dual rank SODIMM.

Hmm, interesting, I tried to bypass that issue by using a clock of 400Mhz like the following picture
[image]

Could you try with the following config in the attached file?
[MIG.tar.gz](https://github.com/eugene-tarassov/vivado-risc-v/files/10150086/MIG.tar.gz)

> If you post details on how you boot qemu, I will add this flow to the project.

I can make a pull request in a bit with a readme and everything :)

gnodipac886, 2022-12-05:
> Merged

I added another PR #122 to add QEMU to the README section

> Linux would not boot without this changes, wait until I commit it.

Please let me know when this happens :), it will be very useful!

eugene-tarassov, 2022-12-07:
I have committed changes to support larger memory modules on VC707 and KC705 boards. For example, VC707 with 8GB SODIMM:

```
make BOARD=vc707 MEMORY_SIZE=0x200000000 CONFIG=rocket64b2 bitstream
```

I looked into slow boot problem. It is caused by newer version of systemd package in the Debian rootfs.  Debian upgraded the package from 250.4 to 252.2. With a faster RISC-V config, Linux used to boot in 3min, now it takes 8min. With a slower CPU it takes about 1hour. The list of changes between systemd versions is huge. I don't know which one is causing problems.

gnodipac886, 2022-12-08:
> I have committed changes to support larger memory modules on VC707 and KC705 boards. For example, VC707 with 8GB SODIMM:

Thank you for this, however when I am running the above command, I am running into the following issue:
```
ERROR: [DRC REQP-83] IDATAIN_connected: riscv_i/DDR/mig_7series_0/u_riscv_mig_7series_0_0_mig/u_memc_ui_top_axi/mem_intfc0/ddr_phy_top0/u_ddr_mc_phy_wrapper/u_ddr_mc_phy/ddr_phy_4lanes_2.u_ddr_phy_4lanes/ddr_byte_lane_B.ddr_byte_lane_B/ddr_byte_group_io/input_[6].iserdes_dq_.idelay_dq.idelaye2: For DELAY_SRC IDATAIN programming the IDATAIN input pin of IDELAYE2 must be connected.
ERROR: [DRC RTSTAT-2] Partially routed nets: 3 net(s) are partially routed. The problem bus(es) and/or net(s) are riscv_i/DDR/mig_7series_0/u_riscv_mig_7series_0_0_mig/u_memc_ui_top_axi/mem_intfc0/ddr_phy_top0/u_ddr_mc_phy_wrapper/u_ddr_mc_phy/ddr_phy_4lanes_2.u_ddr_phy_4lanes/ddr_byte_lane_B.ddr_byte_lane_B/iserdes_clkdiv, riscv_i/DDR/mig_7series_0/u_riscv_mig_7series_0_0_mig/u_memc_ui_top_axi/mem_intfc0/ddr_phy_top0/u_ddr_mc_phy_wrapper/u_ddr_mc_phy/ddr_phy_4lanes_2.u_ddr_phy_4lanes/ddr_byte_lane_B.ddr_byte_lane_B/oserdes_clk, and riscv_i/DDR/mig_7series_0/u_riscv_mig_7series_0_0_mig/u_memc_ui_top_axi/mem_intfc0/ddr_phy_top0/u_ddr_mc_phy_wrapper/u_ddr_mc_phy/ddr_phy_4lanes_2.u_ddr_phy_4lanes/ddr_byte_lane_B.ddr_byte_lane_B/phaser_in_gen.phaser_in_0.
```

Would you happen to have any ideas?

gnodipac886, 2022-12-08:
> ```
> make BOARD=vc707 MEMORY_SIZE=0x200000000 CONFIG=rocket64b2 bitstream
> ```
I got it to build on Vivado fine using this prj file down here:
However, now I get 
```
RISC-V 64, Boot ROM V3.5
Cannot read BOOT.ELF: Disk I/O error
```
<details>
  <summary>.prj file</summary>
  
```xml
<?xml version="1.0" encoding="UTF-8" standalone="no" ?>
<Project NoOfControllers="1">

<!-- IMPORTANT: This is an internal file that has been generated by the MIG software. Any direct editing or changes made to this file may result in unpredictable behavior or data corruption. It is strongly advised that users do not edit the contents of this file. Re-run the MIG GUI with the required settings if any of the options provided below need to be altered. -->

  <ModuleName>riscv_mig_7series_0_0</ModuleName>

  <dci_inouts_inputs>1</dci_inouts_inputs>

  <dci_inputs>1</dci_inputs>

  <Debug_En>OFF</Debug_En>

  <DataDepth_En>1024</DataDepth_En>

  <LowPower_En>ON</LowPower_En>

  <XADC_En>Disabled</XADC_En>

  <TargetFPGA>xc7vx485t-ffg1761/-2</TargetFPGA>

  <Version>4.2</Version>

  <SystemClock>No Buffer</SystemClock>

  <ReferenceClock>Use System Clock</ReferenceClock>

  <SysResetPolarity>ACTIVE HIGH</SysResetPolarity>

  <BankSelectionFlag>FALSE</BankSelectionFlag>

  <InternalVref>0</InternalVref>

  <dci_hr_inouts_inputs>50 Ohms</dci_hr_inouts_inputs>

  <dci_cascade>0</dci_cascade>

  <Controller number="0">
    <MemoryDevice>DDR3_SDRAM/SODIMMs/MT16KTF1G64HZ-1G6</MemoryDevice>
    <TimePeriod>2500</TimePeriod>
    <VccAuxIO>1.8V</VccAuxIO>
    <PHYRatio>4:1</PHYRatio>
    <InputClkFreq>200</InputClkFreq>
    <UIExtraClocks>0</UIExtraClocks>
    <MMCM_VCO>800</MMCM_VCO>
    <MMCMClkOut0> 1.000</MMCMClkOut0>
    <MMCMClkOut1>1</MMCMClkOut1>
    <MMCMClkOut2>1</MMCMClkOut2>
    <MMCMClkOut3>1</MMCMClkOut3>
    <MMCMClkOut4>1</MMCMClkOut4>
    <DataWidth>64</DataWidth>
    <DeepMemory>2</DeepMemory>
    <DataMask>1</DataMask>
    <ECC>Disabled</ECC>
    <Ordering>Normal</Ordering>
    <BankMachineCnt>4</BankMachineCnt>
    <CustomPart>FALSE</CustomPart>
[... 148 lines cut ...]
    <emrMirrorSelection name="Address Mirroring">Disable</emrMirrorSelection>
    <emrRTT name="RTT (nominal) - On Die Termination (ODT)">RZQ/6</emrRTT>
    <emrPosted name="Additive Latency (AL)">0</emrPosted>
    <emrOCD name="Write Leveling Enable">Disabled</emrOCD>
    <emrDQS name="TDQS enable">Enabled</emrDQS>
    <emrRDQS name="Qoff">Output Buffer Enabled</emrRDQS>
    <mr2PartialArraySelfRefresh name="Partial-Array Self Refresh">Full Array</mr2PartialArraySelfRefresh>
    <mr2CasWriteLatency name="CAS write latency">5</mr2CasWriteLatency>
    <mr2AutoSelfRefresh name="Auto Self Refresh">Enabled</mr2AutoSelfRefresh>
    <mr2SelfRefreshTempRange name="High Temparature Self Refresh Rate">Normal</mr2SelfRefreshTempRange>
    <mr2RTTWR name="RTT_WR - Dynamic On Die Termination (ODT)">Dynamic ODT off</mr2RTTWR>
    <PortInterface>AXI</PortInterface>
       <AXIParameters>
       <C0_C_RD_WR_ARB_ALGORITHM>RD_PRI_REG</C0_C_RD_WR_ARB_ALGORITHM>
       <C0_S_AXI_ADDR_WIDTH>33</C0_S_AXI_ADDR_WIDTH>
       <C0_S_AXI_DATA_WIDTH>512</C0_S_AXI_DATA_WIDTH>
       <C0_S_AXI_ID_WIDTH>4</C0_S_AXI_ID_WIDTH>
       <C0_S_AXI_SUPPORTS_NARROW_BURST>0</C0_S_AXI_SUPPORTS_NARROW_BURST>
       </AXIParameters>
  </Controller>

</Project>
```
  
</details>

eugene-tarassov, 2022-12-09:
I'm currently testing an updated version of the scripts, using 4GB and 8GB modules, VC707 and KC705 boards - these boards have similar DDR3 SODIMM interfaces. 4GB module works fine on both boards, 8GB works on KC705, but Linux freezes during boot on VC707. I'm still investigating. BTW, your prj file assumes 1.35V power, I think it is incorrect. DDR3 on VC707 is powered by 1.5V.

gnodipac886, 2022-12-09:
> these boards have similar DDR3 SODIMM interfaces. 4GB module works fine on both boards, 8GB works on KC705, but Linux freezes during boot on VC707. I'm still investigating

Awesome work!
I really appreciate it!
Please let me know if are any updates or anything I can help with :)

eugene-tarassov, 2022-12-12:
> Please let me know if are any updates

I have committed new version of DDR3 modules support. It seems working fine now. Let me know if there is any problem.

gnodipac886, 2022-12-13:
> Let me know if there is any problem.

Hmm, I'm getting this right now, perhaps something wrong with my OS? I'm going to try building the OS again.

```
Begin: Running /scripts/init-premount ... done.
Begin: Mounting root file system ... Begin: Running /scripts/local-top ... done.
Begin: Running /scripts/local-premount ... done.
Begin: Waiting for root file system ... Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
done.
Gave up waiting for root file system device.  Common problems:
 - Boot args (cat /proc/cmdline)
   - Check rootdelay= (did the system wait long enough?)
 - Missing modules (cat /proc/modules; ls /dev)
ALERT!  UUID=68d82fa1-1bb5-435f-a5e3-862176586eec does not exist.  Dropping to a shell!
```

eugene-tarassov, 2022-12-13:
To support more than 2GB memory, I had to make changes in the SD card driver, so you need to rebuild Linux kernel:
```
make clean
./mk-sd-card
```

gnodipac886, 2022-12-16:
> To support more than 2GB memory, I had to make changes in the SD card driver, so you need to rebuild Linux kernel:

Yep that worked, thanks a ton!

cnSchwarzer, 2023-02-05:
> Debian upgraded the package from 250.4 to 252.2.

Is it possible to partially revert this package?

gnodipac886, 2023-03-09:
Hi @eugene-tarassov, Gemmini has merged my commit to the PE to save resources, we can potentially include this in this repo's flow.
https://github.com/ucb-bar/gemmini/blob/dev/src/main/scala/gemmini/PE.scala

eugene-tarassov, 2023-03-13:
Hi @gnodipac886, I have updated this repo to include your changes in the Gemmini PE.

closed 2023-04-22 by gnodipac886