eugene-tarassov/vivado-risc-v #120: Bare Metal and Linux on arty-a7-100t

Nitheesh1609, 2022-12-04:
Hi, 
I am trying to replicate the repo on an ARTY-A7-100T board.
I ran make CONFIG=rocket64b1 BOARD=arty-a7-100t flash. and it has completed succesfully.
**
[image]
**
Then I tried to run xsdb to download the baremetal boot.elf and check the system. But I do not find any RISC-V on it.
[image]

Am I missing something?

eugene-tarassov, 2022-12-04:
Have you pressed the PROG button on the board to load new bitstream into FPGA?

Nitheesh1609, 2022-12-05:
Thanks, I was able to run the bare-metal in the hello world in the UART terminal.
When I try to use the ./mk-sd-card script, I get a sfdisk cannot open error, even when I add have my SD-card connected.
[image]

eugene-tarassov, 2022-12-05:
The disk is mounted. It must not be mounted for the script to work. Try `sudo umount /dev/sde`

Nitheesh1609, 2022-12-06:
/dev/sde was unmounted but for some reason /dev/sde1 and /dev/sde2 were mounted. I was able to build the sd card image and make the sd card.
But Now when I try to boot linux on it,
[image]
I get this unable to access timeout error when booted.

eugene-tarassov, 2022-12-06:
Check your SD card connection, it should look like this:
[image]

Also, at the moment, we are having problems with Debian Linux taking too long time to boot on a slow designs like Arty A7. You will probably need to change Linux timeouts, see [this comment](https://github.com/eugene-tarassov/vivado-risc-v/issues/119#issuecomment-1337976249)

Nitheesh1609, 2022-12-06:
Yes, when I changed my PmodSD from JA to JB I was able to rub the linux. Yeah It took a long time to boot up. I have a minimal linux kernel (Linux with busybox) a boot.elf file with the bzImage, a initrd and a rootfs.cpio.gz file. Can I directly have a SD card with the boot.elf, from which the linux can boot up or do we need to make any changes in the uboot or anyother to get to boot from my custom boot.elf ?

eugene-tarassov, 2022-12-06:
> I have a minimal linux kernel (Linux with busybox)  a boot.elf file with the bzImage

I'm not sure what you mean. The normal way to boot Linux on RISC-V is to run Supervisor Binary Interface (e.g. OpenSBI), which includes u-boot, then u-boot loads initrd and the kernel image. OpenSBI, u-boot and the kernel need to be patched to include drivers from this repo for UART, SD and Ethernet. mk-sd-card does all that patching, adds initrd and rootfs, and creates bootable SD card.

Nitheesh1609, 2022-12-06:
https://github.com/cnrv/fpga-rocket-chip
I am trying to replicate that idea but had a block in linux or risc-v processor not booting up. Following the linux image generation steps, I built a bootable elf file, I wanted to know If I use that elf file in this repo?

eugene-tarassov, 2022-12-06:
> I built a bootable elf file, I wanted to know If I use that elf file in this repo?

No, you cannot use bootable elf file from that repo - the hardware design, peripherals are not same.

But you can use different initrd and rootfs: build it yourself with busybox, or use other Linux distribution instead of Debian, like Ubuntu or whatever.

Nitheesh1609, 2022-12-07:
Sure will try that!

closed 2022-12-07 by Nitheesh1609