eugene-tarassov/vivado-risc-v #127: Specified PRJ file is not readable

cnSchwarzer, 2023-02-01:
[mig_b.zip](https://github.com/eugene-tarassov/vivado-risc-v/files/10559866/mig_b.zip)

INFO: [IP_Flow 19-234] Refreshing IP repositories
INFO: [IP_Flow 19-1704] No user IP repositories specified
false
ERROR: [IP_Flow 19-3460] Validation failed on parameter 'XML_INPUT_FILE(XML_INPUT_FILE)' for Specified PRJ file is not readable  'w:/home/reito/vivado-risc-v/workspace/rocket64b2/vivado-nexys-video-riscv/nexys-video-riscv.srcs/sources_1/bd/riscv/ip/riscv_mig_7series_0_0/mig_b.prj'
. BD Cell 'DDR/mig_7series_0'
INFO: [IP_Flow 19-3438] Customization errors found on 'DDR/mig_7series_0'. Restoring to previous valid configuration.
INFO: [Common 17-17] undo 'set_property'
ERROR: [Common 17-39] 'set_property' failed due to earlier errors.

    while executing
"rdi::add_properties -dict {CONFIG.BOARD_MIG_PARAM ddr3_sdram CONFIG.MIG_DONT_TOUCH_PARAM Custom CONFIG.RESET_BOARD_INTERFACE Custom CONFIG.XML_INPUT_F..."
    invoked from within
"set_property -dict [ list  CONFIG.BOARD_MIG_PARAM {ddr3_sdram}  CONFIG.MIG_DONT_TOUCH_PARAM {Custom}  CONFIG.RESET_BOARD_INTERFACE {Custom}  CONFIG.XM..."
    (procedure "create_hier_cell_DDR" line 74)
    invoked from within
"create_hier_cell_DDR [current_bd_instance .] DDR"
    (procedure "create_root_design" line 54)
    invoked from within
"create_root_design """
    (file "../../board/nexys-video/riscv-2022.2.tcl" line 802)

    while executing
"source ../../board/${vivado_board_name}/${block_design_tcl}"
    (file "../../vivado.tcl" line 78)

    while executing
"source ../../vivado.tcl"
    (file "W:/home/reito/vivado-risc-v/workspace/rocket64b2/system-nexys-video.tcl" line 8)

cnSchwarzer, 2023-02-02:
It seems there're some extra bytes at the begin of file

eugene-tarassov, 2023-02-02:
I have no idea. It works fine for me. I have checked daily build logs for nexys-video 64b2 2022.2 - no issues.

cnSchwarzer, 2023-02-02:
Re clone and the file are all right now

cnSchwarzer, 2023-02-02:
Is it intended to have these \uFEFF characters?
[image]

cnSchwarzer, 2023-02-02:
And the linux kernel seems not able to checkout on Windows drive because of that `aux.c` (aux filename invalid), any existing solution?

cnSchwarzer, 2023-02-02:
Anyway WSL2 works perfect with Linux Vivado GUI, just install another copy on WSL.

closed 2023-02-02 by cnSchwarzer