eugene-tarassov/vivado-risc-v #128: Custom Instructions

cnSchwarzer, 2023-02-02:
Hi, sorry to bother you!

I want to add my custom instructions in this project, and I found RocketChip has something useful: https://github.com/chipsalliance/rocket-chip/blob/master/src/main/scala/scie/SCIE.scala

But I don't know how to easily integrate to this project, and make everything work, do you have related suggestions?

Thank you!

eugene-tarassov, 2023-02-02:
I think the best way is to create a new repo and include this repo as submodule. In your repo, you can add your own Scala files to extend RocketChip functionality, setup [SBT](https://www.scala-sbt.org/) project, add Makefile, etc. You should not need to modify anything in the submodule, you can just refer to it from SBT and Makefile.

> I also learned something named RoCC, is that the same as Gemmini accelerator? 

RoCC is RocketChip interface for adding processor extensions, Gemmini  is an example of such extension that uses RoCC.

cnSchwarzer, 2023-02-03:
Thanks, I'll try figure it out

cnSchwarzer, 2023-02-03:
Does this sound right?
- Make your repo as a submodule
- Create my own top scala file just like your `/src/main/scala/rocket.scala`, but define class of my own. For example extend one of Rocket32s1/Rocket64b2/etc..
- Create my custom RoCC, and build and new it in scala config
- Create Makefile that control over this repo

eugene-tarassov, 2023-02-03:
Yes.

One more thing: Scala is built by SBT (Scala Build Tool), so you will need SBT config files: `build.sbt`, `project/build.properties` and `project/plugins.sbt`.

cnSchwarzer, 2023-02-03:
Thank you very much

closed 2024-02-13 by eugene-tarassov