eugene-tarassov/vivado-risc-v #131: Timed out waiting for device /dev/disk/by-label/BOOT.

cnSchwarzer, 2023-02-04:
I'm running on board `nexys-video` and everything default. Is it normal?

[image]
[image]

cnSchwarzer, 2023-02-04:
I checked the QEMU and the boot contains boot.elf, etc... So is it mounting itself?

cnSchwarzer, 2023-02-04:
Might related to https://github.com/eugene-tarassov/vivado-risc-v/issues/119#issuecomment-1336507535

eugene-tarassov, 2023-02-04:
Yes, it is related to that comment, udev daemon hits timeout on a slow CPU.
The problem is a regression in the mainstream Linux, I have not seen a solution yet.

Workaround is to increase timeouts in the Linux config, see [this comment](https://github.com/eugene-tarassov/vivado-risc-v/issues/119#issuecomment-1337976249) for the script.

closed 2024-02-13 by eugene-tarassov