eugene-tarassov/vivado-risc-v #135: How to single step debug RISC-V system （opensbi Linux kernel etc）

adofirst2018, 2023-02-11:
I have run linux on nexys video follow this repo， Now， I want to debug RISC-V boot。
can you give me some suggestion?

eugene-tarassov, 2023-02-13:
You can use Xilinx System Debugger, XSCT (aka XSDB): [XSCT Commands](https://docs.xilinx.com/r/en-US/ug1400-vitis-embedded/XSCT-Commands). In particular, [stp command](https://docs.xilinx.com/r/en-US/ug1400-vitis-embedded/stp) does stepping.

For source level debugging, you need to use either `dow` or `memap -file` commands to load symbols.

Alternatively, if you prefer a GUI, you can use either Xilinx Vitis or [Eclipse](https://wiki.eclipse.org/TCF/RISC-V).

adofirst2018, 2023-02-14:
@eugene-tarassov thank you very much，Ｉ try to debug RISC-V boot follow your suggestion.

closed 2023-06-07 by adofirst2018