eugene-tarassov/vivado-risc-v #139: Debian GUI support

0ena, 2023-03-03:
Hi,

I see that network connectivity is supported by the RISC-V implementations of this repo and that the cores run Debian Linux.

I wanted to ask if it is possible to run a GUI version of Debian on the implementations provided.
For example I would like to use the VGA port on my Genesys 2 board to run the GUI of the Debian Linux.
Is that possible?

Thank you in advance for your response and time.

Kind regards,
Nassos

eugene-tarassov, 2023-03-04:
It is possible, but not provided in this repo. Adding such support should not be a big problem.

0ena, 2023-03-04:
Hi @eugene-tarassov ,

thank you for your response.
Any insights on what needs to be added (I assume some type of hw controller) and where to look for?
I am curious to see if it worths my time spent on it.

Thanks!

-Nassos

eugene-tarassov, 2023-03-04:
The simple approach is to add "frame buffer" - a memory region to store a video frame.  For example, see [Xilinx Video Series](https://support.xilinx.com/s/topic/0TO2E000000YNwTWAW/xilinx-video-series?language=en_US&tabset-50c42=2&tabset-2ef67=2) of articles on the topic. I'm sure there are plenty of open source implementations too. I have not tried it myself, so I cannot give more specific recommendations.

0ena, 2023-03-04:
I appreciate your info @eugene-tarassov !
I am closing this thread now.

-Nassos

closed 2023-03-04 by 0ena