eugene-tarassov/vivado-risc-v #144: Arty-a7-100t: Error for running bitstream

L0stMyN4me, 2023-03-24:
We followed the steps of readme completely, firstly we execute the command (make CONFIG=rocket64b2 BOARD=arty-a7-100t vivado-gui )to generate the system arty a7-100t. tcl file,
and then executing the tcl file in Vivado(2020.2) to generate the bd. The first time we did not modify the bd, we directly generated  bitstream.
Error reported: Insufficient LUT resources. Subsequently, we deleted some IPs, such as ethernet, and then generated bitstream, but the result was still an error report indicating insufficient resources.
[image]
The development board we use is Arty-a7-100t,vivado version is 2020.2
Could you please tell me which step we did may have  problems？

eugene-tarassov, 2023-03-24:
2-core system is too large for this board - insufficient LUT resources. Use single core configuration: CONFIG=rocket64b1

eugene-tarassov, 2023-03-30:
The project is setup for Digilent Pmod SD adapter. Your adapter has different pinout. You need to modify board/arty-a7-100t/sdc.xdc to match your adapter. Also, make sure you use right socket on the Arty board - socket JB.

ipapal, 2025-04-04:
@L0stMyN4me Hello, have you found a solution?