eugene-tarassov/vivado-risc-v #145: ZCU104: can't pass the synthesis

Xinchen-Fan, 2023-03-27:
I tried to modified the vivado project generated by:

```sh
make CONFIG=rocket64b1 BOARD=nexys-a7-100t bitstream
```

(I followed the instructions in the following issue: [#104 ](https://github.com/eugene-tarassov/vivado-risc-v/discussions/104))

1. change the `ddr2 mig` to `ddr4 mig`
2. delete the `xadc` module
3. modify the constraints.

However, these is only 6% of io resource utilized after the vivado synthesis, can i get some help? thank you.

eugene-tarassov, 2023-03-27:
Why are you concerned about IO resource utilization?
These RISC-V projects use very little of IO resources.
6% is about right.

Xinchen-Fan, 2023-03-27:
Sorry, I mean, the synthesis of Vivado passed while all other resources show a usage of 0%, including lut, lutram, bram and dsp. Vivado has not reported an error

eugene-tarassov, 2023-03-27:
I see. If you are looking at Post-Synthesis Utilization in the Project Summary, it shows only top module utilization, which is useless. To see all utilization numbers, click Report Utilization under Open Synthesized Design. But even that is only an estimate. To see real utilization you need to run implementation.

Xinchen-Fan, 2023-03-27:
> 

Thanks, I'll try it.

EdrisBorne, 2024-05-15:
@xinchen13 were you able to add support for zcu104? I am trying to do the same but find it difficult to adapt it to this repo? May I see the changes you made?

Comet959, 2024-06-28:
> Hello, we find this repo helpful and we want to run Rocket and Gemmini on ZCU-104. I tried to modify the vivado project generated by:
> 
> ```shell
> make CONFIG=rocket64b1 BOARD=nexys-a7-100t bitstream
> ```
> 
> I followed the instructions in the issue [#104 ](https://github.com/eugene-tarassov/vivado-risc-v/discussions/104) and tried:
> 
> 1. change the board_files from `Nexys_A7_100T` to `Zynq UltraScale+ XCZU7EV-2FFVC1156 MPSoC`
> 2. change the `ddr2 mig` to `ddr4 mig`
> 3. delete the `xadc` module
> 4. modify the constraints.
> 
> However, these is only 6% of io resource utilized after the vivado synthesis, can I get some help? thank you.

Hi, are you already running the repo on zcu104? I'm ready to boot rocketchip on my zcu102, can you share your changes based this pro.