eugene-tarassov/vivado-risc-v #147: gcc-8-riscv64-linux-gnu no longer available in newer versions of ubuntu

davekeeshan, 2023-03-28:
When I try and build a new boot image on my recently built version of ubuntu it fails:

```
make[1]: Entering directory '/mnt/sda/projects/vivado-risc-v/u-boot'
/bin/sh: 1: /usr/bin/riscv64-unknown-elf-gcc-8: not found
```

Turns out the apt install of `gcc-8-riscv64-linux-gnu` had failed is no longer available in the standard repo (I missed it on the first pass)

```
Package gcc-8-riscv64-linux-gnu is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
```

This is the release of ubuntu I am on:

```
lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.1 LTS
Release:	22.04
Codename:	jammy
```

(I have another machine running an older version of ubuntu, so am able to proceed, this is just reporting the issue I observed)

kamplianitis, 2023-03-30:
I had a similar issue recently. My ubuntu release is the following.

[image]

Turns out that there is no official support for gcc-8 and consequently there is no official support for the cross-compiler modules for RISCV architectures.

So in order to have a way around the problem I did the following.

```sh
wget http://archive.ubuntu.com/ubuntu/pool/universe/g/gcc-8-cross-ports/gcc-8-riscv64-linux-gnu-base_8.4.0-3ubuntu1cross1_amd64.deb

wget http://mirrors.kernel.org/ubuntu/pool/universe/g/gcc-8/gcc-8_8.4.0-3ubuntu2_amd64.deb

wget http://mirrors.edge.kernel.org/ubuntu/pool/universe/g/gcc-8/gcc-8-base_8.4.0-3ubuntu2_amd64.deb

wget http://mirrors.kernel.org/ubuntu/pool/universe/g/gcc-8/libgcc-8-dev_8.4.0-3ubuntu2_amd64.deb

wget http://mirrors.kernel.org/ubuntu/pool/universe/g/gcc-8/cpp-8_8.4.0-3ubuntu2_amd64.deb

wget http://mirrors.kernel.org/ubuntu/pool/universe/g/gcc-8/libmpx2_8.4.0-3ubuntu2_amd64.deb

 wget http://mirrors.kernel.org/ubuntu/pool/main/i/isl/libisl22_0.22.1-1_amd64.deb

wget http://mirrors.kernel.org/ubuntu/pool/universe/g/gcc-8/libstdc++-8-dev_8.4.0-3ubuntu2_amd64.deb

wget http://mirrors.kernel.org/ubuntu/pool/universe/g/gcc-8/g++-8_8.4.0-3ubuntu2_amd64.deb
```

After downloading all the above files you can install them by running the following commands:

```sh
sudo apt install ./gcc-8-riscv64-linux-gnu-base_8.4.0-3ubuntu1cross1_amd64.deb
sudo apt install ./libisl22_0.22.1-1_amd64.deb ./libmpx2_8.4.0-3ubuntu2_amd64.deb ./cpp-8_8.4.0-3ubuntu2_amd64.deb ./libgcc-8-dev_8.4.0-3ubuntu2_amd64.deb ./gcc-8-base_8.4.0-3ubuntu2_amd64.deb ./gcc-8_8.4.0-3ubuntu2_amd64.deb
sudo apt install ./libstdc++-8-dev_8.4.0-3ubuntu2_amd64.deb ./g++-8_8.4.0-3ubuntu2_amd64.deb
```

If something goes wrong, just try to install each package seperately. After doing the above procedure, you should be good to go.

davekeeshan, 2023-04-03:
@kamplianitis 

Thanks for the reply.  I have done what you have said, and it says that the install is successful.  However I can't find where these tools have been installed to, the are definitely not in the path, and a rudimentary search from `/` down hasn't popped it up for me.

kamplianitis, 2023-04-04:
@davekeeshan 

DEB files just contain the packaged application, along with some identifiers to tell the OS that it's a DEB, and a few instructions for installation. If you open a DEB and browse it, you'll likely find files under the paths where they'll end up once the DEB is installed.

Try doing the following:

1. Open the .deb file with an application like Archive Manager.
2. Inside you will probably see a binary file and two compressed ones. 
3. Search the control zipped file.
4. md5sums will tell you where the .deb installed the package.

kamplianitis, 2023-04-04:
Being an undergraduate student myself, I can tell you that this stuff is kind of new to me too. Without being sure, I think that any `riscv64-*-gcc-8` could work with the precondition that the `gcc-riscv64-unknown-elf` is installed too as you mention. Although, I am skeptical about the fact that driver names of previous or latter versions may install a driver called something else. This could result in failure of vivado running in case the tool searches for the specific driver ( I do not know how likely this can be).  

Anyway, you can always try to run the tools and see whether they're working or not.

davekeeshan, 2023-04-04:
@kamplianitis 

OK time to go whack a mole.  

I used the googles and got to this page, which I always hate when I land here:

[https://packages.ubuntu.com/bionic/gcc-8-riscv64-linux-gnu](https://packages.ubuntu.com/bionic/gcc-8-riscv64-linux-gnu)

After trial and error these were all the extra deb I had to download to my system to get a working `riscv64-linux-gnu-gcc-8` on my system. 

It is always hard now to point to this as being the definitive list (In addition to the ones you have listed above) as some of the required dependencies may have already been installed on my system for some other non riscv reason, but I think the lesson here is when in doubt refer back to the link above and that will tell you what else to get.

```
wget http://mirrors.kernel.org/ubuntu/pool/main/i/isl/libisl19_0.19-1_amd64.deb
wget http://security.ubuntu.com/ubuntu/pool/universe/g/gcc-8-cross-ports/libgcc-8-dev-riscv64-cross_8.4.0-1ubuntu1~18.04cross1_all.deb
wget http://security.ubuntu.com/ubuntu/pool/universe/g/gcc-8-cross-ports/cpp-8-riscv64-linux-gnu_8.4.0-1ubuntu1~18.04cross1_amd64.deb
wget http://security.ubuntu.com/ubuntu/pool/universe/g/gcc-8-cross-ports/gcc-8-riscv64-linux-gnu-base_8.4.0-1ubuntu1~18.04cross1_amd64.deb
wget http://security.ubuntu.com/ubuntu/pool/universe/g/gcc-8-cross-ports/gcc-8-cross-base-ports_8.4.0-1ubuntu1~18.04cross1_all.deb
wget http://security.ubuntu.com/ubuntu/pool/universe/g/gcc-8-cross-ports/gcc-8-riscv64-linux-gnu_8.4.0-1ubuntu1~18.04cross1_amd64.deb
```
```
sudo apt install ./libisl19_0.19-1_amd64.deb
sudo apt install ./libgcc-8-dev-riscv64-cross_8.4.0-1ubuntu1~18.04cross1_all.deb
sudo apt install ./gcc-8-riscv64-linux-gnu-base_8.4.0-1ubuntu1~18.04cross1_amd64.deb
sudo apt install ./cpp-8-riscv64-linux-gnu_8.4.0-1ubuntu1~18.04cross1_amd64.deb
sudo apt install ./gcc-8-cross-base-ports_8.4.0-1ubuntu1~18.04cross1_all.deb
sudo apt install ./gcc-8-riscv64-linux-gnu_8.4.0-1ubuntu1~18.04cross1_amd64.deb
```
However this I think is just a work around to the issue, I still think ultimately the issue is closable when an apt install is enough to to get ones system ready to compile the `u-boot` (but additionally this may be `u-boot`s problem to address not this project)

davekeeshan, 2023-04-04:
Oh I also had to install this when I tried to compile:

`sudo apt install libssl-dev`

because of this:

`fatal error: openssl/evp.h: No such file or directory`

eugene-tarassov, 2023-04-09:
I reworked u-boot patches. Now it builds with the latest gcc, gcc-8 is not needed.

closed 2023-04-09 by eugene-tarassov