eugene-tarassov/vivado-risc-v #154: repo linux-stable unlinked

mbelda, 2023-04-19:
Hello,

I am trying to boot an vc707 with an sd card and I am configuring the sd following mk-sd-image and mk-sd-card.
But on the script [mk-sd-image](https://github.com/eugene-tarassov/vivado-risc-v/blob/master/mk-sd-image#L36) when trying to check the kernel version.
It seems to not recognize the submodule since it is an empty folder and it is not linked on the github web interface.

Thank you in advance.

eugene-tarassov, 2023-04-19:
The script expects a version tag on the currently selected commit in the linux-stable repo. I do see such tag in my copy of the repo. I guess you modified linux-stable to point to a commit which does not have a tag. You can either add a tag to that commit
or select (checkout) a tagged commit or modify the script to assign KERNEL_VER with something that looks like a kernel version.

mbelda, 2023-04-19:
Oh, ok. I was n another version because I modified some files to solve some problems.
But if I go back to a release version I have some issues executing `make linux` on `vivado-risc-v/linux-stable`.
The first error I get is about a dependency library.
~~~
make -I/usr/include/x86_64-linux-gnu/ -C linux-stable ARCH=riscv CROSS_COMPILE=/home/majo/chipyard/.conda-env/riscv-tools/bin/riscv64-unknown-linux-gnu-  oldconfig
make[1]: Entering directory '/home/majo/vivado-rv-vc707/vivado-risc-v/linux-stable'
#
# No change to .config
#
make[1]: Leaving directory '/home/majo/vivado-rv-vc707/vivado-risc-v/linux-stable'
make -I/usr/include/x86_64-linux-gnu/ -C linux-stable ARCH=riscv CROSS_COMPILE=/home/majo/chipyard/.conda-env/riscv-tools/bin/riscv64-unknown-linux-gnu-  all
make[1]: Entering directory '/home/majo/vivado-rv-vc707/vivado-risc-v/linux-stable'
  HOSTCXX scripts/gcc-plugins/latent_entropy_plugin.so
In file included from /home/majo/chipyard/.conda-env/riscv-tools/bin/../lib/gcc/riscv64-unknown-linux-gnu/9.2.0/plugin/include/gcc-plugin.h:28,
                 from scripts/gcc-plugins/gcc-common.h:7,
                 from scripts/gcc-plugins/latent_entropy_plugin.c:78:
/home/majo/chipyard/.conda-env/riscv-tools/bin/../lib/gcc/riscv64-unknown-linux-gnu/9.2.0/plugin/include/system.h:687:10: fatal error: gmp.h: No such file or directory
  687 | #include <gmp.h>
      |          ^~~~~~~
compilation terminated.
make[3]: *** [scripts/gcc-plugins/Makefile:54: scripts/gcc-plugins/latent_entropy_plugin.so] Error 1
make[2]: *** [scripts/Makefile.build:466: scripts/gcc-plugins] Error 2
make[1]: *** [Makefile:1188: scripts] Error 2
make[1]: Leaving directory '/home/majo/vivado-rv-vc707/vivado-risc-v/linux-stable'
make: *** [Makefile:88: linux-stable/arch/riscv/boot/Image] Error 2
~~~

I have checked that I have gmp installed, so I don't know why the compiler does not fid it.
Anyway I just went to the compiler flags and add my path to the gmp library with the `-I` option and then that error disappears but another one is triggered.
 ~~~
In file included from /home/majo/miniforge3/x86_64-conda-linux-gnu/sysroot/usr/include/features.h:375,
                 from /home/majo/miniforge3/x86_64-conda-linux-gnu/sysroot/usr/include/stdio.h:27,
                 from /home/majo/chipyard/.conda-env/riscv-tools/bin/../lib/gcc/riscv64-unknown-linux-gnu/9.2.0/plugin/include/system.h:46,
                 from /home/majo/chipyard/.conda-env/riscv-tools/bin/../lib/gcc/riscv64-unknown-linux-gnu/9.2.0/plugin/include/gcc-plugin.h:28,
                 from scripts/gcc-plugins/gcc-common.h:7,
                 from scripts/gcc-plugins/latent_entropy_plugin.c:78:
/usr/include/x86_64-linux-gnu/sys/cdefs.h:492:49: error: missing binary operator before token "("
  492 | #if __GNUC_PREREQ (4,8) || __glibc_clang_prereq (3,5)
      |                                                 ^
In file included from scripts/gcc-plugins/gcc-common.h:95:
/home/majo/chipyard/.conda-env/riscv-tools/bin/../lib/gcc/riscv64-unknown-linux-gnu/9.2.0/plugin/include/builtins.h:23:10: fatal error: mpc.h: No such file or directory
   23 | #include <mpc.h>
      |          ^~~~~~~
compilation terminated.
make[3]: *** [scripts/gcc-plugins/Makefile:55: scripts/gcc-plugins/latent_entropy_plugin.so] Error 1
make[2]: *** [scripts/Makefile.build:466: scripts/gcc-plugins] Error 2
make[1]: *** [Makefile:1188: scripts] Error 2
make[1]: Leaving directory '/home/majo/vivado-rv-vc707/vivado-risc-v/linux-stable'
make: *** [Makefile:88: linux-stable/arch/riscv/boot/Image] Error 2
~~~

What can I be missing?
I configured the repository as explained in the Usage section of the readme of this repo.

Thank you very much in advance.

eugene-tarassov, 2023-04-19:
> I have some issues executing make linux on vivado-risc-v/linux-stable

I see at least 2 problems:
1. vivado-risc-v/linux-stable is wrong directory to run make
2. you use GCC from chipyard directory, but this repo, vivado-risc-v, has absolutely nothing to do with chipyard.

This is how you can build Linux disk image using this repo - I tried it just now, and it worked fine:
```
git clone https://github.com/eugene-tarassov/vivado-risc-v.git
cd vivado-risc-v
make apt-install
make update-submodules
./mk-sd-image
```

Also, this should be done using Ubuntu 20 LTS host, other Linux distributions are not supported.

mbelda, 2023-04-20:
I just redirected the gcc to my chipyard riscv toolchain. That does not work?
I can't use the script to configure the sd card because I don't have it on a usb adapter, I connect it directly to the computer so that script does not find the correct device.
I have done the script myself step by step using the material provided on your last release, but Linux seems to not boot.
It gets stucked at this point:
~~~
INIT
CMD0
CMD8
ACMD41
CMD58
CMD16
CMD18
LOADING 0x01e00000B PAYLOAD
LOADING  
BOOT

~~~

I have not used you bitstreams instead I have my own with a Large Boom and 8x8 Gemmini. 
I guess that it is enough to support Debian, is it?

mbelda, 2023-04-20:
Ok, I did make all the rules on the Makefile and now I have generated Linux without any problem.

I will see if my bitstream works fine.
Thanks!

closed 2023-04-20 by mbelda