eugene-tarassov/vivado-risc-v #157: Error booting: fsck not present, so skipping root file system

mbelda, 2023-04-27:
Hi,

I had the debian running on my vc707 but I need to pass some files to the sd filesystem to execute some experiments.
I just add the content I needed to /home/debian/myfolder on the rootfs folder and compress that to use it as rootfs.tar.gz on the mk-sd-card script.

I had some issues with the fdisk command on https://github.com/eugene-tarassov/vivado-risc-v/blob/master/mk-sd-card#L32  about the device being used but finally I wipped the sd and rerun it and worked.

But now the Debian is not booting and the following appears on the uart output:

~~~
Loading, please wait...
Starting systemd-udevd version 252~rc3-2
Begin: Loading essential drivers ... done.
Begin: Running /scripts/init-premount ... done.
Begin: Mounting root file system ... Begin: Running /scripts/local-top ... done.
Begin: Running /scripts/local-premount ... done.
Warning: fsck not present, so skipping root file system
[  192.822947] EXT4-fs (mmcblk0p2): INFO: recovery required on readonly filesystem
[  192.830281] EXT4-fs (mmcblk0p2): write access will be enabled during recovery
[  195.281443] EXT4-fs (mmcblk0p2): recovery complete
[  195.313139] EXT4-fs (mmcblk0p2): mounted filesystem with ordered data mode. Quota mode: disabled.
done.
Begin: Running /scripts/local-bottom ... done.
Begin: Running /scripts/init-bottom ... mount: No such file or directory
mount: invalid option --
done.
mount: No such file or directory
run-init: opening console: No such file or directory
Target filesystem doesn't have requested /sbin/init.
run-init: opening console: No such file or directory
run-init: opening console: No such file or directory
run-init: opening console: No such file or directory
run-init: opening console: No such file or directory
run-init: opening console: No such file or directory
No init found. Try passing init= bootarg.
(initramfs) 
~~~

If I did something wrong, how can I transfer he files I need to the sd?

eugene-tarassov, 2023-04-27:
The board runs Debian with networking, so you can use any of normal Linux ways to transfer files over network: sftp, rsync, NFS, autofs, etc. No need to mess with rootfs.tar.gz. Use apt update / apt install to add Linux packages like compilers, editors, etc.

mbelda, 2023-05-05:
I could finally ssh and use the commands you provided.
Thanks!

closed 2023-05-05 by mbelda