eugene-tarassov/vivado-risc-v #160: Boot in emergency mode

mbelda, 2023-05-04:
Hi, 
I just programed the VC707 with a bitstream of a Rocket core + a Gemmini 16x16 accelerator. 
But when I plug the sd card on and try to boot Linux the output shows the folling message and the subsequent error:

~~~
You are in emergency mode. After logging in, type "journalctl -xb" to view
system logs, "systemctl reboot" to reboot, "systemctl default" or "exit"
to boot into default mode.
Give root password for maintenance
(or press Control-D to continue): 
root@debian:~# exit
exit
Reloading system manager configuration
Failed to reload daemon: Connection timed out
Starting default.target
Failed to start default.target: Connection timed out
[FAILED] Failed to start Helper to …chronize boot up for ifupdown.
See 'systemctl status ifupdown-pre.service' for details.
[DEPEND] Dependency failed for Raise network interfaces.
[  OK  ] Reached target Network.
[  OK  ] Reached target Network is Online.
         Starting Notify NFS peers of a restart...
[  OK  ] Started Notify NFS peers of a restart.
~~~

I noticed that using the provided vc707 bitstream on the releases area this does not happen.

What is happening?

mbelda, 2023-05-09:
I fixed it with a design with 2 rocket cores.

mbelda, 2023-05-23:
I have this problem again when trying to run one boom core. And this time I can't add another core.
Any solution, please?

eugene-tarassov, 2023-05-23:
The timeout problem is addressed in the latest code, you need to rebuild SD card.
To make sure you use the latest code, do a **clean** build:
```
git clone https://github.com/eugene-tarassov/vivado-risc-v.git
cd vivado-risc-v
make update-submodules
./mk-sd-card
```

mbelda, 2023-05-24:
I already updated it when you indicated so on another issue.
~~~
commit 7c4cc7d25ed78221c02e59d83f38e2e011a86a33 (HEAD, tag: v3.5.0)
Author: Eugene Tarassov <eugene@largest.net>
Date:   Fri May 5 19:41:48 2023 -0700

    Release v3.5.0
~~~

mbelda, 2023-06-01:
As said on another issue https://github.com/eugene-tarassov/vivado-risc-v/issues/164 it was a problem with my sd-card-image.
Solved now.

closed 2023-06-01 by mbelda