eugene-tarassov/vivado-risc-v #161: Cycle hardware counter

mbelda, 2023-05-17:
Hello,

Is there any special config that must be specified on the design to enable the cycle hardware counter?
I find that in the provided riscv-vc707 bitstream cycles are read correctly but on my designs (Rocket64b2 as it is in `rocket.scala`) I can't read cycles.

eugene-tarassov, 2023-05-17:
The provided riscv-vc707 bitstream is built using Rocket64b2 config from rocket.scala and Vivado 2021.1. There should be no difference.

mbelda, 2023-05-26:
The lecture is not right in any architecture. Probably because of what it is said on Issue https://github.com/eugene-tarassov/vivado-risc-v/issues/163. 
Maybe we can change to that issue and close this.

closed 2023-05-26 by mbelda