eugene-tarassov/vivado-risc-v #163: read clock cycles fail

jack4041313, 2023-05-26:
HI,
I want to read  clock cycle by function "read_cycles();"
This function uis work on spike simulation, like this:
[image]

but when I run file-linux on genesys2, the clock cycle can't be calculated,
function "read_cycles();" always return same value
[image]

eugene-tarassov, 2023-05-26:
It would help if you tell how you implemented read_cycles().
Assuming it uses rdcycle instruction, I don't think rdcycle is supported in Linux.
Standard Linux way to get execution time is to use clock_gettime(), and it appears working fine:
```
debian@debian:~/cycles$ cat t.c
#include <stdio.h>
#include <unistd.h>
#include <time.h>

unsigned read_cycles(void) {
    unsigned cycles;
    asm volatile ("rdcycle %0" : "=r" (cycles));
    return cycles;
}

double read_clock(void) {
    struct timespec t;
    clock_gettime(CLOCK_MONOTONIC, &t);
    return t.tv_sec + t.tv_nsec / 1000000000.0;
}

int main(int argc, char ** argv) {
    unsigned i;
    for (i = 0; i < 100; i++) {
        printf("%12u %12.6f\n", read_cycles(), read_clock());
        sleep(1);
    }
    return 0;
}
debian@debian:~/cycles$ gcc t.c
debian@debian:~/cycles$ ./a.out
   970601647  2679.823193
   970601647  2680.826266
   970601647  2681.827457
   970601647  2682.830341
   970601647  2683.831535
   970601647  2684.833388
   970601647  2685.834598
   970601647  2686.835807
   970601647  2687.837559
   970601647  2688.838744
   970601647  2689.839941
   970601647  2690.842109
   970601647  2691.843297
   970601647  2692.844487
   970601647  2693.845688
   970601647  2694.846878
^C
debian@debian:~/cycles$

```

jack4041313, 2023-05-26:
but I have successfully read clock cycle before,
and now, I want to get conv2d clock cycles 
like this:
[image]

how can I do in debian?

jack4041313, 2023-05-31:
I changed the image on the SD card, it can work fine and I don't know why

@mbelda, I can share image with you if you need.

monniaux, 2023-06-07:
Same problem here. I have three RISC-V machines.

1.  An FPGA running an older version of Vivado-RISC-V with a dual Rocket core.
2. A SiFive U740
3. An FPGA running a recent version of Vivado-RISC-V with a BOOM core.

On 1 and 2, the `rdcycle` instruction works well from Linux userland. On 3, it returns a constant value.

elialaz, 2024-12-29:
@monniaux how do solved this? I facing the same problem

elialaz, 2024-12-30:
@jack4041313 can you share the image please?

elialaz, 2024-12-31:
Just a quick update for everyone: it seems that it is not a software problem but a hardware one. I used a more recent version of OpenSBI for the image build process, and in the rocket chip/qemu the instruction worked fine. I simulated the BOOM core using Chipyard and got the same results, so there is a misconfiguration with this repository BOOM version.