eugene-tarassov/vivado-risc-v #165: Varium C1100 Board support

Chiwawachiwawa, 2023-05-29:
Hello sir, I want to use this project on Varium C1100. This is the first time I use a similar project, but I have encountered many strange problems. I am very presumptuous. Will there be support for this development board?

eugene-tarassov, 2023-05-29:
I don't plan to add support for Varium C1100.
However, the board appears similar to VCU1525, so you can use VCU1525 board files as a starting point for Varium C1100.

Chiwawachiwawa, 2023-05-31:
Thank you very much for your reply sir, I would like to ask you the second question, may I ask if the scope of my modification will go beyond the ./board/C1100 folder? (if I do not modify the ethernet module or other stuffs.

eugene-tarassov, 2023-05-31:
No, you should not need to modify anything beyond the board/C1100 folder.

You can consider adding C1100 into [board/rocket-freq](https://github.com/eugene-tarassov/vivado-risc-v/blob/master/board/rocket-freq) to get higher CPU clock frequency. The default frequencies are rather conservative, C1100 should be able to run faster then that.

Chiwawachiwawa, 2023-05-31:
Thank you sir!

Chiwawachiwawa, 2023-06-04:
> No, you should not need to modify anything beyond the board/C1100 folder.
>
> You can consider adding C1100 into [...]

Hello sir, Sorry for bothering you again,how can I get the original codes of Rocket64z1 produced by RocketChip, if I get it, I can add external devices by myself!

eugene-tarassov, 2023-06-04:
> how can I get the original codes of Rocket64z1 produced by RocketChip

I'm not sure what you mean by "the original codes".
The Rocket64z1 is a config for the Berkeley Out-of-Order Machine (BOOM), it is not produced by RocketChip.
The BOOM repository is checked out in generators/riscv-boom folder.
The BOOM is work in progress, its architecture is extremally complex and not 100% stable yet.
I don't recommend to use it unless you are an expert in BOOM.