eugene-tarassov/vivado-risc-v #166: Add kernel configs

kamplianitis, 2023-06-01:
I am trying to add some more kernel configs to your OS system that is based on the RISCV CPU for the VC-707. My thought process was to  regenerate the linux.config (via xconfig) in the patches folder as I have seen that this afterwards is becoming the .config but when running `./mk-sd-image` the config restarts and the modules do not apply. Is any specific setting/changes that should be done too in order to be successfull? Please can you describe the procedure in order to have some additional modules? 

Thank you for your time in advance

eugene-tarassov, 2023-06-01:
Yes, you can regenerate or edit linux.config to add kernel features.
You probably just need to run a clean build:
```
make clean-submodules linux
```

kamplianitis, 2023-06-02:
After doing the recommended action the same procedure happens and in the end the modules/configs that needed still do not get implemented in the os. This is the message that get popped after cleaning the submodules and replacing the config.

[image]

after completing the process even though the image is generated correctly, still the desired modules are missing

kamplianitis, 2023-06-02:
As it seemed there was an error in cleaning the submodules. After cloning a raw image and implementing the changes the proposed solution worked fine. Thank you for the answer.

closed 2023-06-02 by kamplianitis