eugene-tarassov/vivado-risc-v #168: Baremetal Program

joji-jose, 2023-06-08:
Hi Eugene,
    we were running a program (our own hello.c) ,we came across following error.

[image]

Kindly let us know if the  provided screenshot is not opening.

joji-jose, 2023-06-08:
hi eugene, 
      how to run driver code (fpga-axi-eth.c) ?.Is this the correct driver code for compilation?

eugene-tarassov, 2023-06-08:
> we were running a program (our own hello.c) ,we came across following error

The processor is in S mode. Probably because it is already running Linux.
Are trying to load bare metal app on top of Linux?
Of course it will not work, and it does not make any sense to do that.
A bare metal app needs to be loaded instead of Linux, right after you reset the board and the CPU is in M mode.

> how to run driver code (fpga-axi-eth.c) ?.Is this the correct driver code for compilation?

This is a Linux kernel driver. To run it, you build and boot Linux.
For bare metal you have to write your own drivers.

joji-jose, 2023-06-09:
hi eugene,
         1. how  I build and run  linux kernel driver?
         2. what is the difference between fpga-axi-eth.c and drivers/net/ethernet/xilinx/xilinx_axienet_main.c ?

joji-jose, 2023-06-09:
hi eugene,
       1 . how I built & boot linux kernal driver? 
       2 . what is the difference between fpga-axi-eth.c and drivers/net/ethernet/xilinx/xilinx_axienet_main.c