eugene-tarassov/vivado-risc-v #170: Cannot connect debugger with core using external memory smaller than 0x8000_0000

leduchuybk, 2023-06-20:
I created a Rocket Config with 256 MB external memory. On Address Editor of Vivado project, I map MEM_AXI4 port with Mig ddr on address 0x8000_0000 to 0x8fff_ffff. However, when connecting to openocd, the core informs `Segmentation fault (core dumped)` error.
Can you explain the error?
I saw you split the MEMORY_ADDR_RANGE64 into 2 sections in Makefile. I suppect this cause the issue. 
Can you explain?
Thank you!!!

eugene-tarassov, 2023-06-20:
By default, CPU address 0x8000_0000 is translated to address 0 on MEM_AXI4 bus.
You can change the memory start address by setting RAM_ADDR_OFFSET_MB parameter on the RocketChip wrapper.

leduchuybk, 2023-06-20:
I changed RAM_ADDR_OFFSET_MB = 2048
Then set MEM_AXI4 Bus at 0x8000_0000
However, same result. OpenOCD tells that: `Debug Module did not become active`

eugene-tarassov, 2023-06-20:
In general, this is not the best place to find info on OpenOCD. This project is setup an tested to work with Xilinx Vivado/Vitis, including Xilinx debugger.

closed 2023-07-03 by leduchuybk