eugene-tarassov/vivado-risc-v #172: ./mk-sd-card error

hparpinel, 2023-06-29:
Hi,

when I try the command ./mk-sd-card I get the error message below. 
Copy disk image to:
Disk /dev/sda: 59.49 GiB, 63864569856 bytes, 124735488 sectors
Disk model: MassStorageClass
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x00000000

Device     Boot Start       End   Sectors  Size Id Type
/dev/sda1       32768 124735487 124702720 59.5G  7 HPFS/NTFS/exFAT
Are you sure? [y/N] y
24000+0 records in
24000+0 records out
1572864000 bytes (1.6 GB, 1.5 GiB) copied, 0.607903 s, 2.6 GB/s
Error: Partition(s) 1, 2 on /dev/sda have been written, but we have been unable to inform the kernel of the change, probably because it/they are in use.  As a result, the old partition(s) will remain in use.  You should reboot now before making further changes.

~Hellen

closed 2023-06-29 by hparpinel

zhaokkkkkkkkk, 2023-08-08:
Hello, my project has the same error, how did you solve it，what is the impact of this error

Billkouts96, 2024-03-21:
@hparpinel how you were able to resolve this ? I am getting similar error when trying to inform kernel about the partitions.
Thanks in advance 
Bill