eugene-tarassov/vivado-risc-v #174: Pointers on how to run baremetal with libraries

ewwachter, 2023-07-04:
Hi Eugene,

Many thanks for the project, I've been able to use it with my students on a couple of projects successfully last year.

I was wondering how much effort and if you have any pointers or documentation to help me move to a baremetal code with access to more libraries, such as printf, strcmp, so that I can easily compile benchmarks such as dhrystone on baremetal?

The example you have on the baremetal folder is simple but it makes difficult to include larger programs. I'm in no way complaining :)  just asking if you have any documentation to help in this direction.

Thanks,

eugene-tarassov, 2023-07-05:
No, sorry, at the moment, I don't have any recommendations for baremetal apps.

leduchuybk, 2023-07-11:
You can try this: https://github.com/sifive/freedom-e-sdk
They write bare metal board support package.
Howevere, you must change configuration file