eugene-tarassov/vivado-risc-v #178: linux-stable kernel download problem

ccnuktd, 2023-07-24:
Hi,

When I run make update-submodules, the download of the linux kernel is so slow that I choose to download from the mirror site.

After I confirm that the downloaded content of the mirror station and the original station are correct, I use `git checkout eceb0b1` to switch to correct version.

[image]

But error shows.
```
ccnuktd@LAPTOP-TNKPRGBT:~/linux-stable$ git checkout eceb0b1
error: pathspec 'eceb0b1' did not match any file(s) known to git.
```
Is linux-stable @ eceb0b1 on the project home page correct?

ccnuktd, 2023-07-24:
I re-execute make update-submodules, I still need to download some files. The download speed of these files is so slow that it may take my days to download the updated content. What I am confused about is that I downloaded the latest Linux kernel from the mirror site, why do I still need to download other content here.
Here is some message.
```
ketted@ketted:~/Desktop/vivado-risc-v$ make update-submodules
rm -rf workspace/patch-*-done
git submodule sync --recursive
为 'ethernet/verilog-ethernet' 同步子模组 url
为 'generators/gemmini' 同步子模组 url
为 'generators/gemmini/software/gemmini-rocc-tests' 同步子模组 url
为 'generators/gemmini/software/libgemmini' 同步子模组 url
为 'generators/gemmini/software/onnxruntime-riscv' 同步子模组 url
为 'generators/riscv-boom' 同步子模组 url
为 'generators/sifive-cache' 同步子模组 url
为 'generators/testchipip' 同步子模组 url
为 'linux-stable' 同步子模组 url
为 'opensbi' 同步子模组 url
为 'rocket-chip' 同步子模组 url
为 'u-boot' 同步子模组 url
git -c submodule.torture.update=none -c submodule.software/gemmini-rocc-tests.update=none -c submodule.software/onnxruntime-riscv.update=none submodule update --init --force --recursive
子模组路径 'ethernet/verilog-ethernet'：检出 'f3d5e7452703a25cb394e124bbba648e0c4cf69a'
子模组路径 'generators/gemmini'：检出 'be2e9f26181658895ebc7ca7f7d6be6210f5cdef'
略过子模组 '../../generators/gemmini/'
略过子模组 '../../generators/gemmini/'
子模组路径 'generators/gemmini/software/libgemmini'：检出 '4be220794cfdb834e8ecc2ee7becdf8632cc268c'
子模组路径 'generators/riscv-boom'：检出 '1a153d4974977a2d5f8baa9b4de51b3337d230fe'
子模组路径 'generators/sifive-cache'：检出 'dd1ecafc0c7573a7056ef0946750af31a30a97c1'
子模组路径 'generators/testchipip'：检出 '1952231569c939a9a5e47fa2eef8168405d0136d'

remote: Enumerating objects: 2024853, done.
remote: Counting objects: 100% (139279/139279), done.
remote: Compressing objects: 100% (5002/5002), done.
接收对象中： 0%(11600/2024853), 9.02 MiB | 10.00/KiB/s

ccnuktd, 2023-07-24:
I find the problem. I download linux-kernel but we need linux-kernel stable.
There is a faster mirror site. 
`https://mirrors.tuna.tsinghua.edu.cn/help/linux-stable.git/`

arkountos, 2023-07-24:
I have the same issue, where `make update-submodules` hangs while cloning into `.../vivado-risc-v/linux-stable`.
I noticed that the `linux-stable` folder on the repo isn't actually pointing to a different repo, maybe this is the problem?

eugene-tarassov, 2023-07-24:
linux-stable folder is cloned from git.kernel.org, which is the official repository for Linux kernel. If git.kernel.org accsess is too slow at your location, you can redirect git to a mirror by creating .gitconfig file in your home directory, with lines like these:
```
[url "[https://mirrors.tuna.tsinghua.edu.cn/help/linux-stable.git/"]
        insteadOf = https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git
```