eugene-tarassov/vivado-risc-v #179: How to use other Linux-stable version?

hwirys, 2023-08-04:
Thank you for making a great development tool.
The only one i want to change is Linux version with using latest version of generators(ex. BOOM, rocket-chip generators, ...)
I want to use past version(Linux stable-5.7.1).

So tried these methods, but they are all failed.

1. Change the version of vivado-risc-v
-> Tags v.2.2.0, But initrd & rfs is removed from your relases, so i can't build OS system.
2. Change linux-stable folder
-> The difference between older & newer patch make errors.
3. Build Hardware in Newest Version(Tags: Master) & Build OS in Older version(Tags: v.3.0.0)
-> The difference in bootloader makes error.(v3.0.0 using riscv-pk, and after 3.1.0 using OpenSBI)

Tags: v3.1.0's Linux-stable is so different with 5.7.1, so i can't change it.

Do you have any advice for changing Linux Kernel Version of using it?

eugene-tarassov, 2023-08-04:
You cannot just change version of a component and expect it to work - there will be incompatibilities. I think the best way to change Linux version is your #2: checkout v5.7.1 in the linux-stable folder, then re-work patches to fix all errors.

hwirys, 2023-08-05:
Thank you for your advice.
But rfs(root file system) & initrd is invalid in your github releases.
So When i boot Linux, it stucked at "Starting Kernel..."
How could i get or make rfs & initrd for 5.7.1(or vivado-risc-v v2.2.0)

eugene-tarassov, 2023-08-05:
rootfs & initrd have no dependencies on the kernel version. They should work with any (stable) kernel version. Anyway, I don't build rootfs & initrd, I download them from the Debian website.

> But rfs(root file system) & initrd is invalid in your github releases.

What do you mean? They work fine for me.

hwirys, 2023-08-06:
I think these files have dependencies on kerenel version. 
Because operations in Makefile which debian-riscv64/initrd and debian-riscv64/rootfs.tar.gz are trying to download rootfs.tar.gz & initrd files via vivado-risc-v github. And its site also change when vivado-risc-v version change.

I tried these steps to use linux-kernel v5.7.1
0. Program FPGA image with same version of vivado-risc-v.
1. Go to linux-stable & git checkout v5.7.1
2. Replace fpga-axi-eth.c, fpga-axi-sdc.c, linux.config, linux.patch for 5.7.1
(Most instruction is copied from vivado-risc-v 2.2.0)
3. run ./mk-sd-card (There are no errors)

But, it stucked at "Starting Kernel..."
I changed some lines in linux.config for checking Kernel, but they don't show any message.

Changed line:
"CONFIG_CONSOLE_LOGLEVEL_DEFAULT=15
CONFIG_CONSOLE_LOGLEVEL_QUIET=15
CONFIG_MESSAGE_LOGLEVEL_DEFAULT=7"

[image]

Would i change any other sources like ethernet folder or otherelse?

eugene-tarassov, 2023-08-06:
> Program FPGA image with same version of vivado-risc-v.

I'm not sure what you mean by "same version". I assume it is the latest version.

> Replace fpga-axi-eth.c, fpga-axi-sdc.c, linux.config, linux.patch for 5.7.1

No, you cannot use old drivers (fpga-axi-eth.c, fpga-axi-sdc.c) with latest hardware design.
The latest hardware needs latest drivers.

> But, it stucked at "Starting Kernel..."

Your kernel crashed. Probably because of bad drivers.

hwirys, 2023-08-07:
#I'm not sure what you mean by "same version". I assume it is the latest version.
Yes I program FPGA in the latest version.
And, i re-generated hardware with latest drviers.
But it still hangs at Starting Kerenel...

I re-installed all of codes and i just changed three files/folders.(linux-stable -> Kernel V.5.7.1, linux.patch,  linux.config)
Is there anything else do i need to change in linux.patch or .config?

closed 2024-06-19 by hwirys