eugene-tarassov/vivado-risc-v #183: Support For Alveo U280

MrJimbo2002, 2023-08-14:
Hi Mr.Eugene, many thanks for your such detailed open-source project sharing.

I am new to FPGA and Vivado and please accept my apologies asking not deep questions:

I want to port your U250 design to U280 but encountered the problems below:
[image]
[image]
[image]

I have checked with the documents of QSFP28 Interface, but not sure how to modify the LOC parameter of "dict" as N4, N3 ... AU22 listed in top.xdc.

In terms of the Memory Core Error, I attempted to grep riscv_i/DDR/ddr4_0 and seems that there is no indication of SRL(Super Regional Logic block) and hence no I/O banks for each SLR were found.

Could you help me to debug the above critical warnings and error messages?

Thank you very much in advance for your time commitment and help~

eugene-tarassov, 2023-08-14:
>  not sure how to modify the LOC parameter...

You can find this and other board info in the board documentation. You need to request access to the [Alveo-Vivado lounge](https://www.xilinx.com/member/alveo-vivado.html) and download all U280 docs from there. In particular, you need alveo-u280-xdc and au280_boardfiles - look for U280 / Getting Started / Vivado Design Flow.

Then you carefully review all files you have copied from U250 directory and modify them to match U280 docs.

It also would help to try to create a few simple Vivado designs for the board to learn more about both Vivado and the board.

MrJimbo2002, 2023-08-15:
Hi Mr.Eugene,

Thank you so much for your prompt reply and help. I have added both the alveo-u280 xdc files and au280_boardfiles into my server tools directory under Vivado Xilinx BoardStore folder.

However, when I attempted resynthesing and reimplementing the bitstream generation, the following error messages still displayed:

[image]

[image]

[image]

Apart from basic configuration settings of board parts and function name, I additionally changed CONFIG.C0_CLOCK_BOARD_INTERFACE to be sysclk1 ( originally default_300mhz_clk0) in riscv-2022.2.tcl and CONFIG.CHANNELs to be X0Y44 in ethernet-u280.tcl. I have checked with Alveo U280 Data Sheet and our ethernet.xdc constraint file for QSFP28 Interfaces but failed to find specific channel ports to be used. Both sysclk1 and X0Y44 was suggested according to the tcl command line:

[image]

Could you help me to debug the I/O interface problems at the last stage of Design Rule Check for generation of bitstream file? 

Thank you very much again for your time commitment and assistance. Hope you enjoy a nice day~

eugene-tarassov, 2023-08-15:
> Could you help me to debug the I/O interface problems

You need to look for pin constraints in the alveo-u280 xdc files and and use the info to modify ethernet.xdc.

> sysclk1 and X0Y44 was suggested

These look like correct settings for QSFP1 lane 0 - pins like MGTYRXN0_135 in the XDC file.

MrJimbo2002, 2023-08-21:
I have figured it out. Thank you so much Mr.Eugene~

Nicole-H-u, 2024-01-10:
Hi, Mr Jimbo.
 I have added both the alveo-u280 xdc files and au280_boardfiles, but I still meet Critical Warnings and Memory Core Error you met. Could you share how to resolve them?
Thank you so much for your help.

Nicole-H-u, 2024-01-10:
@MrJimbo2002