eugene-tarassov/vivado-risc-v #184: riscv64 core, possible H extension?

DasAbhi, 2023-08-20:
Hello I was wondering, is it feasible to add the "H" extension for hypervisors to the 64bit core? Or would that require too much re-engineering of your existing design ? 

I am asking because hardware with H extension isn't available for consumers right now, so I am exploring an fpga option. Just wanted to know if it is even possible to add H extension in your riscv64 core.

eugene-tarassov, 2023-08-21:
The hypervisor extension is available.
To enable it, just add `new WithHypervisor` to any of the 64bit core configs.
I personally have not tried it.

FrancoAuri96, 2023-10-08:
@DasAbhi Did you manage to sythetize the core with the H-extension?

Comet959, 2024-06-21:
@DasAbhi Hey, did you finish adding H extension to the repo?

Comet959, 2024-06-26:
Adding `new WithHypervusir ++` to src/main/scala/rocket.scala can finish it.