eugene-tarassov/vivado-risc-v #189: Slow boot on arty-a7-100t

RogerGagnon1956, 2023-09-19:
Everything seems to work with booting Linux on the Arty A100T but the booting process is very at some point (see the attached file). It does not seems to be normal even with such a slow board and to be related to network.
[Boot process.pdf](https://github.com/eugene-tarassov/vivado-risc-v/files/12666343/Boot.process.pdf)
I ran a speed test (speedtest-cli) to evaluate the speed of download/upload and I get  **0.04Mbit/s** for download and **0.87Mbit/s** for upload.
My Internet connection is 120Mbits download and 20MBits upload.
[Network speed.pdf](https://github.com/eugene-tarassov/vivado-risc-v/files/12666419/Network.speed.pdf)
Any idea why this is so slow?
Thanks!

eugene-tarassov, 2023-09-19:
The networking service is blocked by udev daemon taking very long time to start, about 50min. This is unrelated to the network speed. About a your ago, udev became very slow after version update, taking more than 10 times longer to start. I have not seen an explanation for that.

RogerGagnon1956, 2023-09-19:
So, during your test with the Arty A7-100T, you have experienced the same slow speed when booting? This is to confirm that the speed I'm getting is normal.
Thanks!

eugene-tarassov, 2023-09-19:
Not exactly normal, but yes, it is known problem with Linux udev package.