eugene-tarassov/vivado-risc-v #193: Can't install GCC

huahdgithub, 2023-10-22:
**sources.list is the default:**
`deb [arch=riscv64] http://deb.debian.org/debian-ports unstable main
deb [arch=riscv64] http://deb.debian.org/debian-ports unreleased main`

**When doing: sudo apt install gcc, I saw the message:**
`The following additional packages will be installed:
  binutils binutils-common binutils-riscv64-linux-gnu cpp cpp-12 fontconfig-config fonts-dejavu-core gcc-12 libabsl20220623 libaom3 libasan8
  libavif15 libbinutils libbrotli1 libc-dev-bin libc-devtools libc6-dev libcc1-0 libcrypt-dev libctf-nobfd0 libctf0 libdav1d6 libde265-0
  libdeflate0 libfontconfig1 libfreetype6 libgav1-1 libgcc-12-dev libgd3 libgomp1 libheif1 libisl23 libjbig0 libjpeg62-turbo liblerc4 libmpc3
  libmpfr6 libnsl-dev libpng16-16 libsvtav1enc1 libtiff6 libtirpc-dev libwebp7 libx11-6 libx11-data libx265-199 libxau6 libxcb1 libxdmcp6
  libxpm4 libyuv0 linux-libc-dev manpages manpages-dev rpcsvc-proto
Suggested packages:
  binutils-doc cpp-doc gcc-12-locales cpp-12-doc gcc-multilib make autoconf automake libtool flex bison gdb gcc-doc gcc-12-doc glibc-doc
  libgd-tools man-browser
The following NEW packages will be installed:
  binutils binutils-common binutils-riscv64-linux-gnu cpp cpp-12 fontconfig-config fonts-dejavu-core gcc gcc-12 libabsl20220623 libaom3
  libasan8 libavif15 libbinutils libbrotli1 libc-dev-bin libc-devtools libc6-dev libcc1-0 libcrypt-dev libctf-nobfd0 libctf0 libdav1d6
  libde265-0 libdeflate0 libfontconfig1 libfreetype6 libgav1-1 libgcc-12-dev libgd3 libgomp1 libheif1 libisl23 libjbig0 libjpeg62-turbo
  liblerc4 libmpc3 libmpfr6 libnsl-dev libpng16-16 libsvtav1enc1 libtiff6 libtirpc-dev libwebp7 libx11-6 libx11-data libx265-199 libxau6
  libxcb1 libxdmcp6 libxpm4 libyuv0 linux-libc-dev manpages manpages-dev rpcsvc-proto
0 upgraded, 56 newly installed, 0 to remove and 0 not upgraded.
Need to get 49.2 MB of archives.
After this operation, 181 MB of additional disk space will be used.
Do you want to continue? [Y/n]`

**After confirming, I got these error messages:**
Ign:1 http://deb.debian.org/debian-ports unstable/main riscv64 manpages all 6.03-2

Ign:2 http://deb.debian.org/debian-ports unstable/main riscv64 binutils-common riscv64 2.40-2
...
Ign:56 http://deb.debian.org/debian-ports unstable/main riscv64 manpages-dev all 6.03-2
Ign:1 http://deb.debian.org/debian-ports unstable/main riscv64 manpages all 6.03-2
Ign:2 http://deb.debian.org/debian-ports unstable/main riscv64 binutils-common riscv64 2.40-2
...
Ign:56 http://deb.debian.org/debian-ports unstable/main riscv64 manpages-dev all 6.03-2
Ign:1 http://deb.debian.org/debian-ports unstable/main riscv64 manpages all 6.03-2
Ign:2 http://deb.debian.org/debian-ports unstable/main riscv64 binutils-common riscv64 2.40-2
...
Ign:56 http://deb.debian.org/debian-ports unstable/main riscv64 manpages-dev all 6.03-2

**Followed by 56 error messages:**
`Err:1 http://deb.debian.org/debian-ports unstable/main riscv64 manpages all 6.03-2
  Temporary failure resolving 'deb.debian.org'

Err:2 http://deb.debian.org/debian-ports unstable/main riscv64 binutils-common riscv64 2.40-2
  Temporary failure resolving 'deb.debian.org'
...
Err:56 http://deb.debian.org/debian-ports unstable/main riscv64 manpages-dev all 6.03-2
  Temporary failure resolving 'deb.debian.org'`

**As well as several more messages:**
E: Failed to fetch http://deb.debian.org/debian-ports/pool/main/m/manpages/manpages_6.03-2_all.deb  Temporary failure resolving 'deb.debian.'

E: Failed to fetch http://deb.debian.org/debian-ports/pool-riscv64/main/b/binutils/binutils-common_2.40-2_riscv64.deb  Temporary failure res'
...
E: Failed to fetch http://deb.debian.org/debian-ports/pool/main/m/manpages/manpages-dev_6.03-2_all.deb  Temporary failure resolving 'deb.deb'
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?

eugene-tarassov, 2023-10-22:
It works fine for me. Perhaps the error message (Temporary failure ...) is correct for a change, and this is indeed a temporary network failure.

huahdgithub, 2023-10-24:
> It works fine for me. Perhaps the error message (Temporary failure ...) is correct for a change, and this is indeed a temporary network failure.

OK. I realized I forgot to connect my board to the Internet. Now I need to figure out how to configure the IP, gateway, etc. Anyway, thank you very much!