eugene-tarassov/vivado-risc-v #200: Build fails starting from commit 964ac04.

newinnovations, 2023-12-11:
Build fails starting from commit 964ac04.

Build command: `make MAX_THREADS=8 CONFIG=rocket64x1 BOARD=genesys2 bitstream`. Problem occurs with other configs as well.

Error:

```
... SNIP ...
[success] Total time: 29 s, completed Dec 11, 2023, 9:45:29 AM
java -Xmx12G -Xss8M  -Dsbt.io.virtual=false -Dsbt.server.autostart=false -jar /home/embryonic/riscv/sbt-launch.jar assembly
[info] welcome to sbt 1.3.13 (Ubuntu Java 11.0.21)
[info] loading settings for project riscv-build from plugins.sbt ...
[info] loading project definition from /home/embryonic/riscv/project
[info] loading settings for project vivado from build.sbt ...
[info] loading settings for project targetutils from build.sbt ...
[info] loading settings for project gemmini from build.sbt ...
[info] loading settings for project testchipip from build.sbt ...
[info] loading settings for project hardfloat from build.sbt ...
[info] set current project to vivado (in build file:/home/embryonic/riscv/)
[info] 350 file(s) merged using strategy 'Discard' (Run the task at debug level to see the details)
[info] 28 file(s) merged using strategy 'First' (Run the task at debug level to see the details)
[error] java.nio.file.InvalidPathException: Malformed input or input contains unmappable characters: shapeless/$tilde$qmark$greater$?.class
[error]         at java.base/sun.nio.fs.UnixPath.encode(UnixPath.java:145)
[error]         at java.base/sun.nio.fs.UnixPath.<init>(UnixPath.java:69)
[error]         at java.base/sun.nio.fs.UnixFileSystem.getPath(UnixFileSystem.java:279)
[error]         at java.base/java.nio.file.Path.of(Path.java:147)
[error]         at java.base/java.nio.file.Paths.get(Paths.java:69)
[error]         at sbtassembly.Assembly$.$anonfun$reportConflictsMissedByTheMerge$1(Assembly.scala:652)
[error]         at sbtassembly.Assembly$.$anonfun$reportConflictsMissedByTheMerge$4(Assembly.scala:661)
[error]         at scala.collection.TraversableLike.$anonfun$flatMap$1(TraversableLike.scala:245)
[error]         at scala.collection.Iterator.foreach(Iterator.scala:941)
[error]         at scala.collection.Iterator.foreach$(Iterator.scala:941)
[error]         at scala.collection.AbstractIterator.foreach(Iterator.scala:1429)
[error]         at scala.collection.IterableLike.foreach(IterableLike.scala:74)
[error]         at scala.collection.IterableLike.foreach$(IterableLike.scala:73)
[error]         at scala.collection.AbstractIterable.foreach(Iterable.scala:56)
[error]         at scala.collection.TraversableLike.flatMap(TraversableLike.scala:245)
[error]         at scala.collection.TraversableLike.flatMap$(TraversableLike.scala:242)
[error]         at scala.collection.AbstractTraversable.flatMap(Traversable.scala:108)
[error]         at sbtassembly.Assembly$.$anonfun$reportConflictsMissedByTheMerge$3(Assembly.scala:660)
[error]         at scala.collection.parallel.AugmentedIterableIterator.flatmap2combiner(RemainsIterator.scala:133)
[error]         at scala.collection.parallel.AugmentedIterableIterator.flatmap2combiner$(RemainsIterator.scala:130)
[error]         at scala.collection.parallel.immutable.ParVector$ParVectorIterator.flatmap2combiner(ParVector.scala:66)
[error]         at scala.collection.parallel.ParIterableLike$FlatMap.leaf(ParIterableLike.scala:1074)
[error]         at scala.collection.parallel.Task.$anonfun$tryLeaf$1(Tasks.scala:53)
[error]         at scala.runtime.java8.JFunction0$mcV$sp.apply(JFunction0$mcV$sp.java:23)
[error]         at scala.util.control.Breaks$$anon$1.catchBreak(Breaks.scala:67)
[error]         at scala.collection.parallel.Task.tryLeaf(Tasks.scala:56)
[error]         at scala.collection.parallel.Task.tryLeaf$(Tasks.scala:50)
[error]         at scala.collection.parallel.ParIterableLike$FlatMap.tryLeaf(ParIterableLike.scala:1070)
[error]         at scala.collection.parallel.AdaptiveWorkStealingTasks$WrappedTask.internal(Tasks.scala:170)
[error]         at scala.collection.parallel.AdaptiveWorkStealingTasks$WrappedTask.internal$(Tasks.scala:157)
[error]         at scala.collection.parallel.AdaptiveWorkStealingForkJoinTasks$WrappedTask.internal(Tasks.scala:440)
[error]         at scala.collection.parallel.AdaptiveWorkStealingTasks$WrappedTask.compute(Tasks.scala:150)
[error]         at scala.collection.parallel.AdaptiveWorkStealingTasks$WrappedTask.compute$(Tasks.scala:149)
[error]         at scala.collection.parallel.AdaptiveWorkStealingForkJoinTasks$WrappedTask.compute(Tasks.scala:440)
[error]         at java.base/java.util.concurrent.RecursiveAction.exec(RecursiveAction.java:189)
[error]         at java.base/java.util.concurrent.ForkJoinTask.doExec(ForkJoinTask.java:290)
[error]         at java.base/java.util.concurrent.ForkJoinPool$WorkQueue.topLevelExec(ForkJoinPool.java:1020)
[error]         at java.base/java.util.concurrent.ForkJoinPool.scan(ForkJoinPool.java:1656)
[error]         at java.base/java.util.concurrent.ForkJoinPool.runWorker(ForkJoinPool.java:1594)
[error]         at java.base/java.util.concurrent.ForkJoinWorkerThread.run(ForkJoinWorkerThread.java:183)
[error] (assembly) java.nio.file.InvalidPathException: Malformed input or input contains unmappable characters: shapeless/$tilde$qmark$greater$?.class
[error] Total time: 6 s, completed Dec 11, 2023, 9:45:47 AM
make: *** [Makefile:233: workspace/rocket64x1/system-genesys2/RocketSystem.fir] Error 1
```

Logging from previous commit 7d17a9a:

```
... SNIP ...
[success] Total time: 79 s (01:19), completed Dec 11, 2023, 9:25:03 AM
cd rocket-chip && java -Xmx12G -Xss8M  -Dsbt.io.virtual=false -Dsbt.server.autostart=false -jar /home/embryonic/riscv/rocket-chip/sbt-launch.jar assembly
[info] welcome to sbt 1.3.13 (Ubuntu Java 11.0.21)
[info] loading settings for project rocket-chip-build from plugins.sbt ...
[info] loading project definition from /home/embryonic/riscv/rocket-chip/project
[warn] There may be incompatibilities among your library dependencies; run 'evicted' to see detailed eviction warnings.
[info] loading settings for project rocketchip from build.sbt ...
[info] loading settings for project hardfloat from build.sbt ...
Using addons:
[info] set current project to rocketchip (in build file:/home/embryonic/riscv/rocket-chip/)
[warn] There may be incompatibilities among your library dependencies; run 'evicted' to see detailed eviction warnings.
[warn] Multiple main classes detected.  Run 'show discoveredMainClasses' to see the list
[info] Strategy 'discard' was applied to 17 files (Run the task at debug level to see details)
[info] Strategy 'filterDistinctLines' was applied to a file (Run the task at debug level to see details)
[info] Strategy 'rename' was applied to 4 files (Run the task at debug level to see details)
[success] Total time: 15 s, completed Dec 11, 2023, 9:25:41 AM
rm workspace/bootrom.img
... SNIP ...
```

eugene-tarassov, 2023-12-11:
That commit requires a clean build. Either start with `git clone` or clean existing workspace like this:
```
make clean clean-submodules update-submodules
make MAX_THREADS=8 CONFIG=rocket64x1 BOARD=genesys2 bitstream
```

newinnovations, 2023-12-12:
Thanks, but each build is super clean as I execute it within Docker. However there is an issue with `sbt assembly` and Docker, which leads to the error above. Not a problem with this project, will close this issue.

closed 2023-12-12 by newinnovations