eugene-tarassov/vivado-risc-v #201: How to set U280 QSFP pins？

Nicole-H-u, 2024-01-11:
Hi, @eugene-tarassov 
I am trying to use U280, but the QSFP pins of U280 is very different from those of U250(according to alveo-u280-xdc, u280 does not have qsfp0_fs[1:0], qsfp0_resetl, qsfp0_refclk_reset, qsfp0_modsell, and qsfp0_lpmode). I am not sure how to modify the constraints.
Here is the error I met:
First, I met error
> [DRC NSTD-1] Unspecified I/O Standard: 6 out of 157 logical ports use I/O standard (IOSTANDARD) value 'DEFAULT', instead of a user assigned specific value. This may cause I/O contention or incompatibility with the board power or connectivity affecting performance, signal integrity or in extreme cases cause damage to the device or the components to which it is connected. To correct this violation, specify all I/O standards. This design will fail to generate a bitstream unless all logical ports have a user specified I/O standard value defined. To allow bitstream creation with unspecified I/O standard values (not recommended), use this command: set_property SEVERITY {Warning} [get_drc_checks NSTD-1].  NOTE: When using the Vivado Runs infrastructure (e.g. launch_runs Tcl command), add this command to a .tcl file and add that file as a pre-hook for write_bitstream step for the implementation run. Problem ports: qsfp0_fs[1:0], qsfp0_resetl, qsfp0_refclk_reset, qsfp0_modsell, and qsfp0_lpmode.
[DRC UCIO-1] Unconstrained Logical Port: 6 out of 157 logical ports have no user assigned specific location constraint (LOC). This may cause I/O contention or incompatibility with the board power or connectivity affecting performance, signal integrity or in extreme cases cause damage to the device or the components to which it is connected. To correct this violation, specify all pin locations. This design will fail to generate a bitstream unless all logical ports have a user specified site LOC constraint defined.  To allow bitstream creation with unspecified pin locations (not recommended), use this command: set_property SEVERITY {Warning} [get_drc_checks UCIO-1].  NOTE: When using the Vivado Runs infrastructure (e.g. launch_runs Tcl command), add this command to a .tcl file and add that file as a pre-hook for write_bitstream step for the implementation run.  Problem ports: qsfp0_fs[1:0], qsfp0_resetl, qsfp0_refclk_reset, qsfp0_modsell, and qsfp0_lpmode.
Actually, I have tried to delete the undefined qsfp0_fs[1:0], qsfp0_resetl, qsfp0_refclk_reset, qsfp0_modsell, and qsfp0_lpmode, but it seems causing other errors
[image]

Leo-Z-Li, 2024-01-24:
i'm also trying to do this and you can refer to this: https://github.com/eugene-tarassov/vivado-risc-v/issues/62

Nicole-H-u, 2024-02-27:
@Leo-Z-Li Thank you so much for your answer!  But now I am stuck by UART garbled code, though I have modified the board code of u250 for u280 following #62 . Have you ever encountered this problem, or can you give some advice？Thank you very much!

Leo-Z-Li, 2024-02-27:
I'm working on migrating this project onto U55C, but I just got the board and haven't tried it yet. will let you know if I find any clue.

munozher, 2024-05-02:
@Nicole-H-u Hello, I am also having the issue with the wierd uart output. Did you manage to solve it? My guess is that is a main clk issue (porting the design to u280 and u55c for example)

Nicole-H-u, 2024-05-06:
I am using U280, and I found the problem is not related to the soft core,  but due to the some unknown problems on the board, so just flash the memory the uart will be normal. Or email me at 1nicolehu1@gmail.com

tahir2018, 2025-01-22:
Hello, can someone who implemented design on U280 please comment, how to handle IIC interface issue, as IIC interface is not available on board level in u280 unlike u250. 
For U280 following error is reported:
set IIC [ create_bd_cell -type ip -vlnv xilinx.com:ip:axi_iic:2.1 IIC ]
/IIC
  set_property -dict [ list \
   CONFIG.IIC_BOARD_INTERFACE {iic_main} \
   CONFIG.USE_BOARD_FLOW {true} \
 ] $IIC
ERROR: [IP_Flow 19-3461] Value 'iic_main' is out of the range for parameter 'IIC Board Interface(IIC_BOARD_INTERFACE)' for BD Cell 'IIC' . Valid values are - Custom
INFO: [IP_Flow 19-3438] Customization errors found on 'IIC'. Restoring to previous valid configuration.
INFO: [Common 17-17] undo 'set_property'
ERROR: [Common 17-39] 'set_property' failed due to earlier errors.

For U250, no error reported:
# Create instance: IIC, and set properties
  set IIC [ create_bd_cell -type ip -vlnv xilinx.com:ip:axi_iic:2.1 IIC ]
/IIC
  set_property -dict [ list \
   CONFIG.IIC_BOARD_INTERFACE {iic_main} \
   CONFIG.USE_BOARD_FLOW {true} \
 ] $IIC