eugene-tarassov/vivado-risc-v #203: Question about porting to HBM memory

Leo-Z-Li, 2024-02-02:
Hi Mr. Eugene,
i am trying to port this project to alveo u55c. One major change would be the memory because u55c only has HBM memory.  I have a question regarding the C0_DDR4_AXI_CTRL signal as it appears to be used for ECC purposes. My question is why this signal needs to be connected to the qdma pcie, and can I simply remove this connection (marked yellow) because HBM does not have such an interface? 
any suggestions will be greatly appreciated:)
[image]

eugene-tarassov, 2024-02-02:
The signal allows access to memory controller registers from RISC-V or the host CPU. It can be used, for example, for diagnostics. It is not needed to run Linux. You can remove it.

Leo-Z-Li, 2024-02-02:
Thank you for the clarify!

closed 2024-02-03 by Leo-Z-Li

Leo-Z-Li, 2024-04-16:
@munozher Hi, I do have a preliminary design but I haven't tested it on board yet because my board is broken and waiting for AMD to replace it. I can share the design with you for test if you are interested.