eugene-tarassov/vivado-risc-v #206: Help needed using a different version of rocket

SteBruening, 2024-02-09:
Hello I am a student using chipyard v1.10.0 for thesis. 

I am using a modified rocket-core branched from version [rocket-chip @ 47f7b71](https://github.com/chipsalliance/rocket-chip/tree/47f7b7144727f0340d511d35b9f6c7a91b2a276f). I having trouble building a bitstream for the nexys video using this version. 

I was able to build the bitstream with the default rocket versions in your repository v3.5.0 and main. When trying this with my version of rocket I run into sbt errors:

```
[info] Updating 
[info] Resolved  dependencies
[warn] 
[warn]  Note: Unresolved dependencies path:
[error] sbt.librarymanagement.ResolveException: Error downloading org.scalameta:semanticdb-scalac_2.13.10:4.4.32
[error]   Not found
[error]   Not found
[error]   not found: /net/home/bruening/.ivy2/local/org.scalameta/semanticdb-scalac_2.13.10/4.4.32/ivys/ivy.xml
[error]   not found: https://repo1.maven.org/maven2/org/scalameta/semanticdb-scalac_2.13.10/4.4.32/semanticdb-scalac_2.13.10-4.4.32.pom
[error]   not found: https://oss.sonatype.org/content/repositories/snapshots/org/scalameta/semanticdb-scalac_2.13.10/4.4.32/semanticdb-scalac_2.13.10-4.4.32.pom
[error]   not found: https://oss.sonatype.org/content/repositories/releases/org/scalameta/semanticdb-scalac_2.13.10/4.4.32/semanticdb-scalac_2.13.10-4.4.32.pom
[error]   not found: /net/home/bruening/.m2/repository/org/scalameta/semanticdb-scalac_2.13.10/4.4.32/semanticdb-scalac_2.13.10-4.4.32.pom
[error]         at lmcoursier.CoursierDependencyResolution.unresolvedWarningOrThrow(CoursierDependencyResolution.scala:249)
[error]         at lmcoursier.CoursierDependencyResolution.$anonfun$update$35(CoursierDependencyResolution.scala:218)
[error]         at scala.util.Either$LeftProjection.map(Either.scala:573)
[error]         at lmcoursier.CoursierDependencyResolution.update(CoursierDependencyResolution.scala:218)
[error]         at sbt.librarymanagement.DependencyResolution.update(DependencyResolution.scala:60)
[error]         at sbt.internal.LibraryManagement$.resolve$1(LibraryManagement.scala:52)
[error]         at sbt.internal.LibraryManagement$.$anonfun$cachedUpdate$12(LibraryManagement.scala:102)
[error]         at sbt.util.Tracked$.$anonfun$lastOutput$1(Tracked.scala:69)
[error]         at sbt.internal.LibraryManagement$.$anonfun$cachedUpdate$20(LibraryManagement.scala:115)
[error]         at scala.util.control.Exception$Catch.apply(Exception.scala:228)
[error]         at sbt.internal.LibraryManagement$.$anonfun$cachedUpdate$11(LibraryManagement.scala:115)
[error]         at sbt.internal.LibraryManagement$.$anonfun$cachedUpdate$11$adapted(LibraryManagement.scala:96)
[error]         at sbt.util.Tracked$.$anonfun$inputChanged$1(Tracked.scala:150)
[error]         at sbt.internal.LibraryManagement$.cachedUpdate(LibraryManagement.scala:129)
[error] (rocketchip / update) sbt.librarymanagement.ResolveException: Error downloading org.scalameta:semanticdb-scalac_2.13.10:4.4.32
[error]   Not found
[error]   Not found
[error]   not found: /net/home/bruening/.ivy2/local/org.scalameta/semanticdb-scalac_2.13.10/4.4.32/ivys/ivy.xml
[error]   not found: https://repo1.maven.org/maven2/org/scalameta/semanticdb-scalac_2.13.10/4.4.32/semanticdb-scalac_2.13.10-4.4.32.pom
[error]   not found: https://oss.sonatype.org/content/repositories/snapshots/org/scalameta/semanticdb-scalac_2.13.10/4.4.32/semanticdb-scalac_2.13.10-4.4.32.pom
[error]   not found: https://oss.sonatype.org/content/repositories/releases/org/scalameta/semanticdb-scalac_2.13.10/4.4.32/semanticdb-scalac_2.13.10-4.4.32.pom
[error]   not found: /net/home/bruening/.m2/repository/org/scalameta/semanticdb-scalac_2.13.10/4.4.32/semanticdb-scalac_2.13.10-4.4.32.pom
[error] Total time: 3 s, completed Feb 7, 2024, 4:36:47 PM
make: *** [Makefile:211: workspace/rocket64b2/system.dts] Error 1 
```

I figured merging my changes into Chipyard 1.11 and using the new rocket version there would be the best way to resolve this (since it closely matches the version used on main in your repo), however i am unable to migrate to the newer version since it cannot run on the university computers due to the outdated GLIBC version 2.28 installed on them. I dont have any admin rights. 

Can you give me any instructions to make this repo work with  [rocket-chip @ 47f7b71](https://github.com/chipsalliance/rocket-chip/tree/47f7b7144727f0340d511d35b9f6c7a91b2a276f). I only need the bitsteam since I am planning to use the prebuild images for the rest. 

Help would be greatly appreciated.

eugene-tarassov, 2024-02-09:
The error is caused by incompatibility between selected versions of RocketChip and Chisel. The releases of this repo use versions of RocketChip and other components that were carefully selected and thoroughly tested to make sure that they build and work fine. I recommend to rebase your changes to one of those RocketChip versions.

SteBruening, 2024-02-10:
Thank you for your suggestion, i will try that. However then i will have to make that RocketChip version work with Chipyard 1.10.0 aswell, since i need the simulator from Chipyard. 
I found that Chipyard 1.11.0 and your repo use a version of RocketChip only 4 commits different. The problem with that is that Chipyard 1.11.0 requires a newer Version of GLIBC that my Institute Computers don't have. I cannot make any changes to the systems here.

SteBruening, 2024-02-10:
The Chisel version was the problem. I have downgraded to 3.5.6 and was able to build the bitstream with that version. Debain boot was successful on the FPGA. 

Thank you for your work and for pointing me into the right direction.