eugene-tarassov/vivado-risc-v #210: Add support for VCU118

cassebas, 2024-02-28:
Hi Eugene,
At work I can use a VCU118 FPGA. Could you add support for this board?
Or, if you give me some pointers on how to add the necessary files/configuration, I can try to add support myself.
Thnx!

eugene-tarassov, 2024-02-29:
> Could you add support for this board?

No, I don't have the board.

> if you give me some pointers 

You need to add board/vcu118 directory with board support files.
Use other board directories as example, board/vc707 would be most suitable in this case.
Copy files from board/vc707 and changed/rework them for the new board.

peterknoben, 2024-03-08:
@cassebas, We recently got a VCU118. After I get everything working on the genesys2 board I will have a look at making the board files for the VCU118. If that is successful I can share it.

cassebas, 2024-04-24:
I am working on reworking the vc707 board files and make them work with the vcu118. I have a question about the memory interface. I see in the board/vc707 directory several .prj files that I think are generated by the MIG tool within Vivado? I want to try and generate these files for the vcu118, but the memory on the vcu118 is DDR4 instead of DDR3. Is this a problem? For example, the user guide of the vcu118 talks about 80-bits wide interface to the memory modules.

Will the rocket core be able to use DDR4?

eugene-tarassov, 2024-04-25:
> but the memory on the vcu118 is DDR4 instead of DDR3. Is this a problem?

No, it is not a problem.
You can use u200, u250 or vcu1525 boards as example, they also have DDR4.

cassebas, 2025-01-15:
Hi Eugene,
I'm still trying to add support for the vcu118 board. I now have a basic configuration that is at least able to generate a bitstream. I have based the configuration on a combination of the vc707 config and the vcu1525 config. From the latter I have taken (more or less) the DDR4 component.

But I get a warning during the implementation, which is probably important (see below). Do I have to explicitly add the pins of the ddr4 interface port in the xdc? The vc707 uses a different approach, it uses an XML .prj file in which a pin selection is present. In the vcu1525 config I couldn't discover the way that the relevant pins are added to the design.

Also I saw a riscv_wrapper.v that is included in the vcu1525 case, would this also be needed for my setup?
Thanks again! 

```
WARNING: [DRC RTSTAT-10] No routable loads: 123 net(s) have no routable loads.
The problem bus(es) and/or net(s) are
riscv_i/DDR/ddr4_0/inst/u_ddr4_mem_intfc/u_ddr_cal_riu/LMB_CE_riu,
riscv_i/DDR/ddr4_0/inst/u_ddr4_mem_intfc/u_ddr_cal_riu/LMB_UE_riu,
riscv_i/DDR/ddr4_0/inst/u_ddr4_mem_intfc/u_ddr_cal_top/u_ddr_cal/u_xsdb_arbiter/Q[12],
riscv_i/DDR/ddr4_0/inst/u_ddr4_mem_intfc/u_ddr_cal_top/u_ddr_cal/u_xsdb_arbiter/Q[13],
riscv_i/DDR/ddr4_0/inst/u_ddr4_mem_intfc/u_ddr_cal_top/u_ddr_cal/u_xsdb_arbiter/Q[14],
riscv_i/DDR/ddr4_0/inst/u_ddr4_mem_intfc/u_ddr_cal_top/u_ddr_cal/u_xsdb_arbiter/Q[15],
riscv_i/DDR/ddr4_0/inst/u_ddr4_mem_intfc/u_phy2clb_phy_rdy_upp/SYNC[0].sync_reg[1],
riscv_i/DDR/ddr4_0/inst/u_ddr4_mem_intfc/u_phy2clb_fixdly_rdy_low/SYNC[0].sync_reg[1],
riscv_i/DDR/ddr4_0/inst/u_ddr4_mem_intfc/u_phy2clb_fixdly_rdy_upp/SYNC[0].sync_reg[1],
riscv_i/DDR/ddr4_0/inst/u_ddr4_mem_intfc/u_phy2clb_phy_rdy_low/SYNC[0].sync_reg[1],
riscv_i/DDR/ddr4_0/inst/u_ddr4_mem_intfc/u_phy2clb_phy_rdy_upp/SYNC[1].sync_reg[1],
riscv_i/DDR/ddr4_0/inst/u_ddr4_mem_intfc/u_phy2clb_fixdly_rdy_upp/SYNC[1].sync_reg[1],
riscv_i/DDR/ddr4_0/inst/u_ddr4_mem_intfc/u_phy2clb_phy_rdy_low/SYNC[1].sync_reg[1],
riscv_i/DDR/ddr4_0/inst/u_ddr4_mem_intfc/u_phy2clb_fixdly_rdy_low/SYNC[1].sync_reg[1],
riscv_i/DDR/ddr4_0/inst/u_ddr4_mem_intfc/u_phy2clb_fixdly_rdy_low/SYNC[2].sync_reg[1]
... and (the first 15 of 121 listed).
'''

eugene-tarassov, 2025-01-15:
> Do I have to explicitly add the pins of the ddr4 interface port in the xdc?

If your have vcu118 board file in Vivado, you can tell Vivado to use pins from that file instead of xdc.
The related setting looks like this: `CONFIG.C0_DDR4_BOARD_INTERFACE {ddr4_sdram_c0}`
Use appropriate interface name instead of `ddr4_sdram_c0`.

> Also I saw a riscv_wrapper.v that is included in the vcu1525 case, would this also be needed for my setup?

The file is workaround for Vivado inability to handle GTs in a block design.
If you don't use GTs, you don't need the file.

cassebas, 2025-02-10:
Hi Eugene,
I'm still working on getting the vcu118 to run a rocket32s1 core. I was able to generate a bitstream. But when I program the bitstream onto the FPGA, I don't get the result I want. The output on the UART only says ``NAP*..` ``

After having programmed the FPGA, the hardware manager shows me the following:
[image]

When I try to connect to the core with xsdb, I get the following:

```
****** System Debugger (XSDB) v2024.2
  **** Build date : Oct 29 2024-10:16:47
    ** Copyright 1986-2022 Xilinx, Inc. All Rights Reserved.
    ** Copyright 2022-2024 Advanced Micro Devices, Inc. All Rights Reserved.

xsdb% connect
tcfchan#0
xsdb% targets -set -filter {name =~ "Hart #0*"}
no targets found with "name =~ "Hart #0*"". available targets:
  1  xcvu9p
     2  MicroBlaze Debug Module at USER2
     3  Legacy Debug Hub
xsdb%
```

Do you have any idea what could be wrong, or a pointer for how to investigate this problem?
Thanks again!