eugene-tarassov/vivado-risc-v #211: Error: Partition(s) 1, 2 on /dev/loop0 have been written

yyx32, 2024-02-29:
Hello, I have completed the process of setting up the system environment, but when I using the "mk-sd-image.sh" script to set up the SD card, an error message is reported as follows:
```
yyx@ubuntu:~/riscv/projects/vivado-risc-v$ ./mk-sd-image 
make: 'debian-riscv64/initrd' is up to date.
make: 'debian-riscv64/rootfs.tar.gz' is up to date.
1500+0 records in
1500+0 records out
1572864000 bytes (1.6 GB, 1.5 GiB) copied, 0.480678 s, 3.3 GB/s
SD image device: /dev/loop0
Checking that no-one is using this disk right now ... OK

Disk /dev/loop0: 1.48 GiB, 1572864000 bytes, 3072000 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes

>>> Created a new DOS disklabel with disk identifier 0x15c165e7.
/dev/loop0p1: Created a new partition 1 of type 'W95 FAT16 (LBA)' and of size 64 MiB.
/dev/loop0p2: Created a new partition 2 of type 'Linux' and of size 1.4 GiB.
/dev/loop0p3: Done.

New situation:
Disklabel type: dos
Disk identifier: 0x15c165e7

Device       Boot  Start     End Sectors  Size Id Type
/dev/loop0p1 *      2048  133119  131072   64M  e W95 FAT16 (LBA)
/dev/loop0p2      133120 3071999 2938880  1.4G 83 Linux

The partition table has been altered.
Error: Partition(s) 1, 2 on /dev/loop0 have been written, but we have been unable to inform the kernel of the change, probably because it/they are in use.  As a result, the old partition(s) will remain in use.  You should reboot now before making further changes.
```
How can I fix this?

eugene-tarassov, 2024-02-29:
I believe this is Linux kernel bug, and running `sudo apt update && sudo apt upgrade` should fix it.

yyx32, 2024-03-04:
Hi @eugene-tarassov ，I tried the method you mentioned, but there was still the same error after running it.

peterknoben, 2024-03-08:
I am also running into this problem. Updating to the latest Linux kernel did not fix it for me. Do you have any other suggestions?

eugene-tarassov, 2024-03-08:
The script works fine for me. I don't see such errors.

One possible cause of the error is stale (or leaked) file descriptor in the kernel.
In this case rebooting the machine should fix it.

0ena, 2024-03-20:
Hi @eugene-tarassov ,

I am getting the exact same issue as the original poster.
I confirmed that it started the past couple weeks.

I tried the `update & upgrade` solution, as well as the `restart`, but neither of them worked.
Any other workarounds on this?
Can you share your set up if possible?

My machine has the following:

```
OS: Ubuntu 22.04.4 LTS jammy
Kernel: 5.15.0-101-generic
RISC-V GCC version: riscv64-linux-gnu-gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0

```

Thank you very much for your help and time!

Kind regards,
Nassos

eugene-tarassov, 2024-03-20:
I don't know how to get the tools to work on Ubuntu 22.04.4. I'm using Ubuntu 20.04.6 and I don't see such errors.

Billkouts96, 2024-03-21:
Hello . I am encountering the exact same issue. I am using Ubuntu 20.04.6 .
What kernel version you ise @eugene-tarassov ?
I am stuck here for more than one week( have tried all possible workarounds)

Thanks in advance

eugene-tarassov, 2024-03-21:
> What kernel version you use @eugene-tarassov ?

```
eugene@gate:~/projects/risc-v/vivado-risc-v$ cat /etc/issue
Ubuntu 20.04.6 LTS \n \l

eugene@gate:~/projects/risc-v/vivado-risc-v$ uname -a
Linux gate 5.4.0-172-generic #190-Ubuntu SMP Fri Feb 2 23:24:22 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux
eugene@gate:~/projects/risc-v/vivado-risc-v$
```
I tried:
```
make clean update
./mk-sd-image
```
It worked OK.

I also tried Ubuntu 22.04.4 LTS machine, and got the error:
```
Device       Boot  Start     End Sectors  Size Id Type
/dev/loop0p1 *      2048  133119  131072   64M  e W95 FAT16 (LBA)
/dev/loop0p2      133120 3071999 2938880  1.4G 83 Linux

The partition table has been altered.
Error: Partition(s) 1, 2 on /dev/loop0 have been written, but we have been unable to inform the kernel of the change, probably because it/they are in use.  As a result, the old partition(s) will remain in use.  You should reboot now before making further changes.
eugene@smash:~/vivado-risc-v$ cat /etc/issue
Ubuntu 22.04.4 LTS \n \l

eugene@smash:~/vivado-risc-v$ uname -a
Linux smash 5.15.0-100-generic #110-Ubuntu SMP Wed Feb 7 13:27:48 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux
eugene@smash:~/vivado-risc-v$
```

As a work around, you can download SD card image from the releases area of this repo, and use dd to copy it to SD card. It has a bit older Linux kernel - 6.3.12 vs 6.7.9, but it should work just fine.

Billkouts96, 2024-03-22:
Thanks a lot for the quick response @eugene-tarassov !
Your  proposed workaround worked fine. Actually, we downloaded the latest SD card image from the release area and ran ./mk-sd-card skip_mk_img

Rade94, 2024-03-27:
Hello, I had the same problem and managed to fix it by changing two things in mk-sd-image script: 
1) the line 66 to:
$ sudo losetup -fP $SD_IMG

2) replaced sfdisk with fdisk for creating partition:
replace lines 76 to 80 to:
sudo fdisk ${SD_LOOP} << EOF
n
p
1
2048
+64M
t
e
a
n
p
2
133120
3071999
p
w
EOF

eugene-tarassov, 2024-03-28:
According to my tests, adding -P option for losetup does fix the error, while replacing sfdisk with fdisk appears not needed.

radev-vtool, 2024-03-28:
@eugene-tarassov  yes, you are right. The -P option fixes it. Thanks for this remark.

Rade94, 2024-03-28:
@eugene-tarassov yes, you are right. The -P option fixes it. Thanks for this remark.