eugene-tarassov/vivado-risc-v #220: Trying to add our custom bootloader

suyogchaudhary, 2024-04-18:
****Hello sir,
Here we are trying to add our custom bootloader on genesys2 board.We have our elf file. We flashed that elf file on SD card but we are facing following issues.
RISC-V 64, Boot ROM V3.6
Cannot access SD: Timeout
Cannot mount SD: Disk not ready

Guide us on how can be add our own bootloader for it.We also tried to put it into  mcs file but still issue remains the same.
Is there any partucular address where we need to put our elf file? or any other method?

eugene-tarassov, 2024-04-18:
Before trying your custom bootloader, try to build and run provided [bare-metal](https://github.com/eugene-tarassov/vivado-risc-v/tree/master/bare-metal) example. Then use it as a reference.

suyogchaudhary, 2024-04-22:
Thank you for your help. I tried with hellow world its working.But i am nt getting from where to start the writimng of bootloader. How you will approach if you are writing the  bootloader for the same?
i am  trying at add coreboot to this riscv-server Needed your help

suyogchaudhary, 2024-04-25:
Hello 
i used hello world as reference for my custom bootloader. but only blank screen is appearing