eugene-tarassov/vivado-risc-v #223: Makefile:240: workspace/rocket64b2/system.dts] Error 1

sidharthp5792, 2024-04-25:
While building bitstream for nexys video with configuartion of rocket64b1 the following errror has been observed
``Exception: sbt.TrapExitSecurityException thrown from the UncaughtExceptionHandler in thread "run-main-0"
[error] Nonzero exit code: 1
[error] (Compile / runMain) Nonzero exit code: 1
[error] Total time: 129 s (02:09), completed 25-Apr-2024, 12:22:50 PM
make: *** [Makefile:240: workspace/rocket64b2/system.dts] Error 1
``
Is there a resolution for this issue?

eugene-tarassov, 2024-04-25:
This message only indicates that sbt exited because of a fatal error. The original error message should be somewhere else in the build log.

sidharthp5792, 2024-05-01:
Ohh okay , thanks for the reply.

Error got resolved after changing the vivado version to 2020.2

closed 2024-05-01 by sidharthp5792