eugene-tarassov/vivado-risc-v #226: CAN Digilent Pmod IP or Vivado IP?

ahmrr, 2024-05-25:
Hello! Thank you for this wonderful project, Mr. Tarassov. 

I wish to add a CAN peripheral. Would you recommend doing this using the provided [Vivado CANFD IP](https://www.xilinx.com/products/intellectual-property/ef-di-canfd.html) or the Digilent Pmod CAN IP? If you would recommend the Digilent IP, could you please provide some guidance on where each IP pin should be connected in the IO block diagram? Below is an image of Digilent's IP.
[image]

Thanks in advance!

eugene-tarassov, 2024-05-26:
I personally have no tried to add CAN bus.

As I understand, the PmodCAN is not a CAN controller, it is an adapter for Digilent Pmod CAN board. The board has MCP25625 CAN controller with SPI host interface.

The Vivado AXI CAN appears to be true CAN controller, which does not require external controller. It still requires external CAN transceiver IC.

You choose one depending on hardware you going to use. You connect the IP to AXI interconnect in the IO block in the RISC-V block design.

ahmrr, 2024-05-27:
Thank you @eugene-tarassov! My apologies, I did not mention my setup. I am running this SoC (rocket64b1) and Linux on an Arty A7-100T FPGA board and want to incorporate the Digilent Pmod CAN. Vivado's IP is also not free to use, so I have decided to use the Digilent Pmod IP. I believe I have connected the IP to the AXI bus correctly, following the AXI Uartlite example.

As both the Pmod's SPI and GPIO interfaces need to be connected to the AXI bus, I was wondering if you could provide some guidance on what the device tree would look like for both these devices, especially the GPIO device? For now, I have the following, which I am not sure are correct:

```
osc: oscillator {
    compatible = "fixed-clock";
    #clock-cells = <1>;
    clock-frequency  = <100000000>;
    clock-output-names = "osc";
};

/* inside io-block */
 spi0: spi@60040000 {
	reg = <0x60040000 0x10000>;
	is-decoded-cs = <0>;
	num-cs = <1>;
	status = "okay";
    	interrupt-parent = <&{/soc/interrupt-controller@c000000}>;
	interrupts = <4>;
	
	spidev@0 {
	    compatible = "microchip,mcp25625";
	    spi-max-frequency = <10000000>;
	    clocks = <&osc>;
	};
};

gpio0: gpio@60050000 {
	gpio-controller ;
	reg = <0x60050000 0x10000>;
	status = "okay";
    	interrupt-parent = <&{/soc/interrupt-controller@c000000}>;
	interrupts = <5>;
	/* ngpios = <6>;  was experimenting with this */
};
```

For reference, the Pmod IP and related drivers are [here](https://github.com/Digilent/vivado-library/tree/master/ip/Pmods/PmodCAN_v1_0). Sorry for the beginner-level questions; I am very new in this area! The IP for the Pmod CAN is also not very thoroughly documented, so I was hoping for some more experienced guidance.

eugene-tarassov, 2024-05-27:
In Linux, device tree is used to pass configuration info to a device driver. So, the format of the info is defined by the driver. You posted a link to Digilent CAN Pmod docs, but I don't see a Linux driver there. It looks like Digilent provides only bare metal driver. It is impossible to tell the device tree format without a Linux driver. You need to find a suitable Linux driver for the device, or develop one. Then you can read the driver source code to figure out the device tree contents.

BTW, the README has section about [adding a peripheral device](https://github.com/eugene-tarassov/vivado-risc-v/tree/master?tab=readme-ov-file#modding-the-design-optional-adding-a-peripheral-device), but you need a driver before you can use this instructions.

ahmrr, 2024-05-29:
Thank you so much for all your help @eugene-tarassov. I greatly appreciate it!

Qualitatively, how "difficult" do you think it would be to port Digilent's aforementioned baremetal driver to Linux?

To the best of my understanding, another option would be to treat the Pmod as a generic SPI device (and related GPIO device) and write some software to manipulate the CAN controller, right? In this case, since (I believe) the Digilent IP uses Vivado's SPI and GPIO IPs, Xilinx's SPI and GPIO drivers should work?

I suppose this is a question for the folks at Digilent, but Linux provides a [driver for the MCP25625](https://github.com/Xilinx/linux-xlnx/blob/master/drivers/net/can/spi/mcp251x.c), which is the chip used in the Pmod CAN. Do you think this driver would be compatible with the Pmod?

Thanks again!

ahmrr, 2024-07-06:
Thank you for your help @eugene-tarassov! I have gotten the MCP25625 initialized via SPI:

`[    7.322020] mcp251x spi0.0 can0: MCP25625 successfully initialized.`

I am currently trying to get the GPIO row of the Pmod (the one that contains the INT pin) working. I added an entry in the address editor and edited the device tree accordingly, trying to add the GPIO controller and a GPIO interrupt in the SPI part. This is what my device tree additions currently are:

```
osc: oscillator {
    compatible = "fixed-clock";
    #clock-cells = <0>;
    clock-frequency  = <20000000>;
    clock-output-names = "osc";
};

spi0: spi@60040000 {
    compatible = "xlnx,axi-quad-spi-3.2", "xlnx,xps-spi-2.00.a";
    reg = <0x60040000 0x10000>;
    interrupt-parent = <&{/soc/interrupt-controller@c000000}>;
    interrupts = <4>;
    is-decoded-cs = <0>;
    num-cs = <1>;
    status = "okay";

    xlnx,num-ss-bits = <0x1>;
      
    can0: can@0 {
        compatible = "microchip,mcp25625";
        interrupt-parent = <&gpio>;
        interrupts = <0 0x8>; /* GPIO 0, active low */
	spi-max-frequency = <10000000>;
        reg = <0>;
        clocks = <&osc>;
    };
};

gpio: gpio@60050000 {
    compatible = "xlnx,xps-gpio-1.00.a";
    reg = <0x60050000 0x10000>;
    interrupt-parent = <&{/soc/interrupt-controller@c000000}>;
    interrupts = <5>;
    gpio-controller;
    #gpio-cells = <2>;
    interrupt-controller;
    #interrupt-cells = <2>;

    xlnx,all-inputs = <0x0>;
    xlnx,dout-default = <0x0>;
    xlnx,gpio-width = <0x4>;
    xlnx,interrupt-present = <0x0>;
    xlnx,is-dual = <0x0>;
    xlnx,tri-default = <0x2>; /* 0010 */
};
```

But when I tried to set up the CAN bus using `ip link set can0 up type can bitrate 125000 loopback on`, I got:

```
[ 3087.282962] genirq: Setting trigger mode 8 for irq 15 failed (xgpio_set_irq_type+0x0/0xdc)
[ 3087.294134] mcp251x spi0.0: failed to acquire irq 15
```

But the GPIO controller is being detected just fine in `/sys/class/gpio`:
```
export  gpiochip512  unexport
```

In addition, when I try to read the GPIO value of the interrupt pin using sysfs (which is active low, so I think it should be 1 all the time) I get 0.

Do you have any idea why this could be happening? Also, the reason I tried to get GPIO working in the first place is because without using the INT pin, if I only provided an interrupt for the SPI bus, I got the following error instead:
```
[ 2568.332983] mcp251x spi0.0: failed to acquire irq 0
```

Thank you in advance, and thanks again for all your help!

eugene-tarassov, 2024-07-06:
I think you cannot use GPIO module to handle CAN interrupts, but you can connect CAN INT directly to any unused interrupt inputs of the CPU.

ahmrr, 2024-07-09:
Thanks @eugene-tarassov! I wanted to avoid doing it that way, but I found Digilent's Pmod Bridge IP and used it to split the INT wire off the Pmod interface connection and connect it to the interrupt controller. I'm encountering a strange problem related to CAN bus messages. When I send a message, I can only send one (which is successfully received by the second node) before I need to reset the CAN bus. If I don't reset it after sending a message, it isn't received by the other node. Also, no messages are able to be received by the FPGA at all. 

I thought this might have something to do with the fact that the interrupt is active low instead of active high. Do I have to specify this to the interrupt controller somehow, or invert the signal in the Vivado block design?

ahmrr, 2024-07-15:
Thank you for all the help @eugene-tarassov! I have gotten the CAN bus working; the issue had to do with the Pmod IP Digilent provided :)

closed 2024-07-15 by ahmrr