eugene-tarassov/vivado-risc-v #228: What is the correct workflow for 'modding the design' (adding new peripheral)

cassebas, 2024-05-27:
Hi Eugene,
I tried to add my own IP by following the instructions under 'Modding the design (optional): adding a peripheral device'. I first generated a normal bitstream of the Rocket64b2l2 config. Then I opened Vivado with

```make CONFIG=rocket64b2l2 BOARD=genesys2 vivado-gui```

I added my IP to the block design, validated it and synthesized it. Vivado seems happy.
Then I closed Vivado and modified important sources, like e.g. the board tcl files to create pins for the new IP block and connect it all.

Finally, I run
```make CONFIG=rocket64b2l2 BOARD=genesys2 bitstream```

Unfortunately, I first get this error:
```java.nio.file.FileAlreadyExistsException: /home/caspar/local/git/vivado-risc-v/workspace/rocket64b2l2/system-genesys2/RocketSystem.fir```
So I removed this file and ran the make command again.

Now a new error pops up:
```java.nio.file.FileAlreadyExistsException: /home/caspar/local/git/vivado-risc-v/workspace/rocket64b2l2/system-genesys2/RocketSystem.anno.json```
Again I remove this file (oops), perhaps I shouldn't have because now I get:
```Exception in thread "main" firrtl.annotations.AnnotationFileNotFoundException: Annotation file /home/caspar/local/git/vivado-risc-v/workspace/rocket64b2l2/system-genesys2/RocketSystem.anno.json not found!```

Do you know why these errors appear and what I'm doing wrong? I think I cannot simply remove the workspace/rocket64b2l2 directory, because then I will loose my Vivado work I think?

Thanks again!

eugene-tarassov, 2024-05-27:
I'm not sure why sbt now throws `java.nio.file.FileAlreadyExistsException`.
As workaround,  run `rm -rf workspace/rocket64b2l2/system-genesys2`.
This will keep Vivado files, which are stored in `workspace/rocket64b2l2/vivado-genesys2-riscv`.

cassebas, 2024-05-28:
Yes, that worked. My custom IP works now.

Would it be sensible/possible to put the Vivado modifications in git? That would allow me to get back to a certain version using Git, instead of manually making edits in Vivado again.

eugene-tarassov, 2024-05-28:
> Would it be sensible/possible to put the Vivado modifications in git?

Yes. When you modify a design, Vivado shows all modifications as Tcl commands in the Tcl Console view. Save these commands in a file and put the file into git. Alternatively, you can save whole design as a script using `write_bd_tcl` command.

closed 2024-05-29 by cassebas