eugene-tarassov/vivado-risc-v #230: Problems when connecting the Rocketchip core with JTAG interface using BSCANE2 and bscan2jtag module

Caiyujie007, 2024-06-04:
I hope my FPGA design has a JTAG port to debug the Rocketchip core, so I generate the Rocketchip using "CONFIG=rocket64b2j", and the JTAG port can be exposed. And I want to use another FPGA which uses BSCANE2 and 'bscan2jtag.vhdl' to drive the Rocketchip core using JTAG. To avoid potential connection and timing problems, I firstly instantiate the "BSCANE2", "bscan2jtag" and the Rocketchip core in single FPGA. There is an input PIN named 'S_BSCAN_bscanid_en' in module bscan2jtag and I drive this PIN use 'VIO' IP. In addtion, I modified the 'timing-constraints.tcl' to correctly constrain the BSCANE2.TCK as 'tck_pin'.
But I found I cannot successfully find to the Rocketchip device using "XSDB" tool whenever I drive the 'S_BSCAN_bscanid_en' into 0 or 1, how to fix this problem? Thanks.
[image]

eugene-tarassov, 2024-06-04:
You need to use Debug Bridge, which is Vivado wrapper for BSCANE2. Your block design should look like this:
[image]

Caiyujie007, 2024-06-06:
It works when using 'Debug Bridge'. Thank you very much.

closed 2024-06-12 by Caiyujie007