eugene-tarassov/vivado-risc-v #231: Understanding the role of co-simulation

sramichetty20019, 2024-06-06:
Hi, everyone. I am trying to understand the role of Qemu here(as mentioned in the repository). Is it just used to install dependencies, or are we trying something like a co-simulation or running qemu interfaced with Rocket on the PL side?.... I am really new to this. Please help me figure out if my understanding is right. 

We are not doing any co-simulation using qemu in this project, right, we are just using it to install dependencies, and also, we are not running qemu on the PL or PS side or PL+PS

eugene-tarassov, 2024-06-06:
A co-simulation is not supported. QEMU work flow is provided for those who wants to run RISC-V Linux applications without FPGA board.  It is completely optional and provided for convenience.

sramichetty20019, 2024-06-06:
So does qemu emulate the  hardware configuration changes model in rocket along with a riscv os?or is it just for installing dependencies using default qemu-system-riscv64

eugene-tarassov, 2024-06-06:
No, qemu does not emulate the hardware configuration. It uses a generic RISC-V model.
Yes, it can be used to build various software components.

sramichetty20019, 2024-06-06:
Thank you so much for the clarification and instant response  !!! much appreciated

closed 2024-06-06 by sramichetty20019