eugene-tarassov/vivado-risc-v #235: How to use RGMII instead of GMII for the ethernet on BOARD=kc705?

EternitYjl, 2024-06-12:
Hello! Thanks for your repositories first! I have a board=xc7k325tffg900-2(as the same as BOARD=kc705). However, it is RGMII for the ethernet on my board. (I have deleted the ethernet and then successfully finished the project.) So what should I do to add my ethernet part? By the way, my vivado version is 2020.1.

eugene-tarassov, 2024-06-12:
Use genesys2 or sitlinv-xc7k325t board files as example of RGMII.

EternitYjl, 2024-06-13:
Thanks for your reply. I tried to use board=genesys2 to fit my board. (The ethernet core on my board is RTL8211E-VB.) I just changed the number of pins to fit my board in the .xdc files. However, there is something wrong:

`[Vivado 12-1411] Cannot set LOC property of ports, Could not legally place instance rgmii_txc_OBUF_inst at AH27 (IOB_X0Y55) since it belongs to a shape containing instance riscv_i/IO/ethernet_stream_0/inst/rgmii_odelay_clk. The shape requires relative placement between rgmii_txc_OBUF_inst and riscv_i/IO/ethernet_stream_0/inst/rgmii_odelay_clk that can not be honoured because it would result in an invalid location for riscv_i/IO/ethernet_stream_0/inst/rgmii_odelay_clk. ["D:/BoomFPGA/genesys2/mychange/ethernet.xdc":15]
`
and

`[Place 30-58] IO placement is infeasible. Number of unplaced terminals (1) is greater than number of available sites (0).
The following are banks with available pins: 
 IO Group: 0 with : SioStd: LVCMOS18   VCCO = 1.8 Termination: 0  TermDir:  Out  RangeId: 1 IdelayId: 2 Drv: 12  has only 0 sites available on device, but needs 1 sites.
	Term: rgmii_txc
`

I am sure that I do not mix the pins in the ethenet.xdc. And I did not change any timing constraints because I did known how it works.(maybe it is the problem?) Could you please tell me how to solve these problems. 

[image]
[image]

These are my constraints(I did not change the .tcl files). I am hoping for your reply. Thank you very much.

eugene-tarassov, 2024-06-13:
It looks like Vivado is trying to say ODELAY is not available on your board RGMII connection.

RGMII interface depends on the board design, in particular: TXDLY on/off, RXDLY on/off, IO bank type (ODELAY available or not). Total 8 possible combinations. You need to learn how to handle these variations or search internet for an example that matches your board.

The Nexys Video board uses RTL8211E-VB phy without ODELAY, it could be a better match to your board.

EternitYjl, 2024-06-19:
Thanks for your reply. It seems like Nexys Video is better for me. I have tried but failed.
[image]
In the vivado, there are also some problems.
[image]
[image]
I think there should not be this problem `[xilinx.com:ip:smartconnect:1.0-1] riscv_io_axi_s_0: The device(s) attached to /M03_AXI do not share a common clock source with this smartconnect instance.   Re-customize this AXI SmartConnect instance to add a new clock pin and connect it to the same clock source of the IP attached to /M03_AXI to prevent futher clock DRC violations.
`. And I don't know how to solve them.
The part of ethernet has bothered me so long. And for my study, I just need the gcc compiler(riscv64-*-gcc). I have tried to get the gcc-riscv64-linux-gnu（4:13.2.0-7）and its dependency packages from debian.org with sd but still failed So is there a linux system with (riscv64) gcc directly available or whole dependency packages for `[debian-riscv64.sd.img` without ethernet connection? Or do you have some suggestions for me to get gcc complier? Thank you very much.