eugene-tarassov/vivado-risc-v #238: Question about RocketChip MEM_AXI4 to DDR memory

cassebas, 2024-07-15:
Hi Eugene,
The RocketChip component of the riscv block design connects with the DDR component through an AXI4 interface. Within the DDR component, an AXI SmartConnect is used to connect the S00_AXI slave port to the M00_AXI master port, which in turn connects to the generated mig_7series_0 component (interface to the DDR3 on my Genesys2 board).

I'd like to write my own IP that replaces this 'AXI SmartConnect'. Do you know what type of memory that is assumed from the perspective of the RocketChip? In other words, can I somehow know in advance the values of the AWCACHE and ARCACHE signals of the AXI4 interface?

I think it will be easier to support just one type of memory, hopefully non-buffered.

Thnx!

eugene-tarassov, 2024-07-15:
It looks like RocketChip uses 1111 on the AWCACHE and ARCACHE signals for cacheable memory access. Otherwise it is 0000 for data and 0011 for instructions. But you can ignore these signals, a memory controller is not required to support caching or buffering.

cassebas, 2024-07-18:
I have another question, I'm trying to figure out how to implement my own IP that replaces the 'AXI SmartConnect'. But I see some challenges, e.g. I have to cross clock domains so I have to use a FIFO probably. In combination with the control signals, this doesn't seem trivial.

The reason for making my own IP is that I would like to read (real-time) the addresses that are requested by the MEM_AXI4 master. I don't see how I can plug in my own registers between MEM_AXI4 and the S00_AXI from the SmartConnect.

Would there be another way to read/monitor the addresses that are requested on the bus, other than creating my custom SmartConnect bridge?

Thanks again.

eugene-tarassov, 2024-07-18:
Yes, reimplementing AXI interconnect is a big project. It would be much easier to implement a VHDL or Verilog module that sits between MEM_AXI4 and SmartConnect. The module can pass through the AXI bus and read/monitor the addresses. Another option is to add a Scala module to RocketChip to do the same.

cassebas, 2024-07-18:
Ah yes, I was trying to think of something like this, but I was under the assumption that in that case I would introduce an extra Master interface and hence an extra address space?

Is your idea to make a module (VHDL in my case) that has an AXI slave port in and an AXI master port out? So it would actually be a simple AXI interconnect but then without addess width difference and without clock domain crossing?

eugene-tarassov, 2024-07-18:
> I would introduce an extra Master interface and hence an extra address space?

It will, probably, show up in Vivado as address space, but it does not matter.

> Is your idea to make a module (VHDL in my case) that has an AXI slave port in and an AXI master port out?

Yes.

> So it would actually be a simple AXI interconnect but then without address width difference and without clock domain crossing?

Yes. In the module, you can just connect slave to master without additional logic.
BTW, VHDL wrapper generated for RocketChip does similar thing connecting RocketChip memory bus to MEM_AXI4 port.

cassebas, 2024-07-19:
It worked! I added the passthrough like you said, and my program runs fine.
Thanks again!

closed 2024-07-19 by cassebas