eugene-tarassov/vivado-risc-v #240: Kernel boot fail on Arty 100T board :

imocanu, 2024-07-20:
OS system : Ubuntu 22.04 ( Vivado v2024.1 64-bit )
project release : commit cdca76bbeebe8ea02333fd41829ffdf69d21a03c  ( Linux kernel 6.9.6 )
project build : make CONFIG=rocket64b1 BOARD=arty-a7-100t bitstream

..........
Gave up waiting for root file system device.  Common problems:
 - Boot args (cat /proc/cmdline)
   - Check rootdelay= (did the system wait long enough?)
 - Missing modules (cat /proc/modules; ls /dev)
ALERT!  UUID=68d82fa1-1bb5-435f-a5e3-862176586eec does not exist.  Dropping to a shell!
(initramfs) 

Boot log :
[boot_log.txt](https://github.com/user-attachments/files/16321464/boot_log.txt)

One possible issue is the default address range for ddr3 :

[image]

Seems that is a similar issue with boot sequence from : https://github.com/eugene-tarassov/vivado-risc-v/issues/235 

Any idea how can be fixed this boot issue ?

eugene-tarassov, 2024-07-20:
The kernel failed to detect SD card. Perhaps because you are trying to boot kernel 5.19.16, which is about 2 years old. Or the kernel was not built properly, e.g. missing SD card driver. The current supported kernel version is 6.9.6.

imocanu, 2024-07-20:
Thanks for feedback.
boot_log was a quick check for an older kernel.

With latest kernel is the same output : 
[kernel_6.9.6_log.txt](https://github.com/user-attachments/files/16321822/kernel_6.9.6_log.txt)

eugene-tarassov, 2024-07-20:
Then it is some other problem with SD card, like incompatible Pmod SD card adapter.
During boot, the kernel should show messages about SD card like this:
```
[    8.044848] mmc0: host does not support reading read-only switch, assuming write-enable
[    8.067856] mmc0: new high speed SDHC card at address aaaa
[    8.093032] mmcblk0: mmc0:aaaa SL16G 14.8 GiB
```

ipapal, 2025-04-04:
Hi @imocanu, have you found any solution ?