eugene-tarassov/vivado-risc-v #241: Error in Makefile for command: make CONFIG=rocket64b2 BOARD=nexys-video vivado-gui

walrider3, 2024-07-21:
I get the same error when trying to generate the bitstream from the command line. Furthermore, to open the design in Vivado, I don't know which files to include.
**Command:** make CONFIG=rocket64b2 BOARD=nexys-video vivado-gui
**Result:** 
java -Xmx12G -Xss8M  -cp `realpath target/scala-*/system.jar` firrtl.stage.FirrtlMain -i workspace/rocket64b2/system-nexys-video/RocketSystem.fir -o RocketSystem.v --compiler verilog \
  --annotation-file workspace/rocket64b2/system-nexys-video/RocketSystem.anno.json \
  --custom-transforms firrtl.passes.InlineInstances \
  --target:fpga
Exception in thread "main" firrtl.passes.PassExceptions:
firrtl.passes.CheckHighFormLike$ResetInputException:  @[rocket-chip/src/main/scala/devices/debug/Periphery.scala 122:19]: [module RocketSystem] Abstract Reset not allowed as top-level input: debug.reset
firrtl.passes.CheckHighFormLike$ResetInputException:  @[rocket-chip/src/main/scala/devices/debug/Periphery.scala 122:19]: [module RocketSystem] Abstract Reset not allowed as top-level input: debug.clockeddmi.dmiReset
firrtl.passes.PassException: 2 errors detected!
make: *** [Makefile:266: workspace/rocket64b2/system-nexys-video.v] Error 1

eugene-tarassov, 2024-07-21:
Your copy of the repo is damaged. You need to do a clean build, or reapply patches/rocket-chip.patch.

eugene-tarassov, 2024-07-22:
I'm not sure what you mean by "downloaded using the zip". Please, follow instructions in the README. They start with
```
sudo apt install git make
git clone https://github.com/eugene-tarassov/vivado-risc-v.git
```
and so on.

The patch is applied when you run make, see the Makefile for details.

walrider3, 2024-07-23:
I mean I downloaded the repository by going to the upper right part of the repo, clicking on the code option, and downloading the git repo in a zip file.  This command git clone https://github.com/eugene-tarassov/vivado-risc-v.git doesn't work for me since it fails downloading the entire repo. Also, when I download the repo in a zip file, some files are empty as those files reference other git repos, so I download them seperately by clicking on them and going to their page. However, linux-stable leads to a 404 page. So, I downloaded the linux kernel from the official repo and put it in the file. I tried the --recursive but that also doesn't work.